rmcanany/SolidEdgeHousekeeper #6: Change Input Directory from a single path to a list of paths

rmcanany, 2022-12-07:
Suggested by @Monk_The_OCD_Engie.  Would speed up Top Level Assembly searches.

farfilli, 2022-12-09:
In my opinion, the way the list of files is created has to be changed; at the moment the list is reloaded many times during the process.
We need methods to add files to the list: from directory, subdirectory, list or drag&drop
We need a method to remove files from the lists (select and delete key)
We need a method to clean the list (a button)

The listview recently introduced is capable to store the full path of each element so the process file will read the full path from there.

farfilli, 2022-12-10:
This issue will be solved with #15 that I'm working on, you can close it

rmcanany, 2022-12-10:
Will be solved as part of Issue #15

closed 2022-12-10 by rmcanany