rmcanany/SolidEdgeHousekeeper #9: Polish user interface

rmcanany, 2022-12-07:
Suggested by @farfilli 

20221207 In progress

farfilli, 2022-12-09:
Would you consider moving to the SegoeUI font for the whole application?

farfilli, 2022-12-09:
In the last commit, the migration to ListView is complete.
Files found will be grouped per type: Assemblies, parts, sheetmetals and drafts.
To delete the selections in the ListView just press the ESC key.
The unprocessed files will get an empty checkbox, successfully processed files will get a green checkmark, and processed with errors will get an alert mark.

[image]

rmcanany, 2022-12-09:
I missed your question about the font.  That sounds like a good idea.  Hopefully it is not a matter of individually changing the font of every single widget.  I'd like to know how it is done.

Regarding the logo, my wife is a graphic designer.  Would you mind if she took a crack at it?  It looks like Gimp can export a *.ico file.  Is that what you used?

farfilli, 2022-12-10:
Every help is welcome, I just took the SE logo and overlapped a brush. I do use Green fish Icon Editor but GIMP is good as well, just make the icon at least 64x64 32x32 24x24 and 16x16

About the font you just change it in the main form and all the controls will follow it automatically

rmcanany, 2022-12-10:
OK.  We'll play around with the logo.

By the way, feel free to assign yourself to issues you're working on.  Not sure if it earns you any extra "points" on GitHub, but you ought to take credit for your contributions!

farfilli, 2022-12-10:
Am I allowed to do it? I mean, I didn't find the way to assign them

farfilli, 2022-12-10:
> to @farfilli <https://github.com/farfilli>.

About the icons this is a good source, I do use the style Colour

https://icons8.com/app/windows

farfilli, 2022-12-15:
The general page user interface is complete now.

rmcanany, 2022-12-15:
A huge improvement!  Thank you!

closed 2022-12-15 by rmcanany