rmcanany/SolidEdgeHousekeeper #14: ClickOnce Installation

farfilli, 2022-12-09:
It would be good to create a ClickOnce installation that checks the online repository for updates.
In this way for a user would be much easier to install Housekeeper and get automatic updates.

With local resources, it works great and it is the way I distribute software in my network, I have never done it with an online repository but it's sure possible, SeSpy works this way.

rmcanany, 2022-12-09:
Very cool!  I'll read up on it.

farfilli, 2022-12-15:
I followed this tutorial to do it; it seems to work even if sometimes the update at open doesn't trigger, haven't figured out it yet

https://stackoverflow.com/questions/13512029/is-it-possible-to-offer-a-clickonce-installer-on-github

rmcanany, 2022-12-15:
On hold for now.

rmcanany, 2024-01-03:
Closing for now.

closed 2024-01-03 by rmcanany