rmcanany/SolidEdgeHousekeeper #15: Differente way to handle the file list

farfilli, 2022-12-10:
We need to change the way the list it's handled.
I will create a branch to experiment with a new way to populate and handle the list.
The user will decide when reload and what to add and what to remove from the list.
We will have this functionality: Add by Folder (with subfolder optional), add by list (txt, csv, xls), add by drag & drop, delete selected items by del key, clear the whole list, filter by type, process all, process selected, save list (txt, csv, xls)

rmcanany, 2022-12-10:
Sounds good.  You're ambitious!

One thing we need to keep in mind is that the Input Directory serves two purposes at the moment -- it serves as the top level folder for the file list, AND it designates the search scope for 'where used' when processing a 'top level assembly'.  The top level assembly may be somewhere below the top level folder.

There is a request to turn the top level folder into a list.  It was from another user, but would be handy for our workflow as well.

Open to ideas on how to integrate/improve the two uses of the input directory...

farfilli, 2022-12-10:
My idea is to add a section in the list called "Files sources" and then the
modify the search sub to use that items. Items will be single folders and
or subfolders

rmcanany, 2022-12-10:
OK.  I just wanted you to be aware of the search scope functionality.  I'm totally open to a different way of handling it.

farfilli, 2022-12-10:
You will have to test it extensively 😅

farfilli, 2022-12-10:
I created a branch in my repository to introduce this functionality, here a quick update of the status

There is elements that will be delete when completed

At the moment only retrieve from single folder and subfolder works, multiple path already work, the counter need to be updated at the moment reports only the last search.

Import lists and top level asm still not implement, only added buttons

[image]

rmcanany, 2022-12-10:
Wow!

farfilli, 2022-12-11:
Robert I need your help to test it.
I haven't complete the TopLevel assembly processing. Before do this I need to understand how it works. Could you explain me why you search on folders instead of read the links in the asm file?

Probably I misunderstood it's purpose

Thanks

Francesco

P.s. do I need to pull my branch for you to test it or can you directly connect to it?

rmcanany, 2022-12-11:
I do the search on the top level assembly (TLA) one of two ways.  

One, called "bottom up" does the standard recursion on the assembly, doing 'where used' for every new link it hits.  It is efficient when many files in the search scope are not related to the TLA.  It should be fast on an indexed drive, but we don't have one so I'm not positive that's the case.

The other, called 'top down', builds a graph of the links in a folder tree by inspecting each file's links.  It then traverses the graph starting with the TLA.  Top down is a lot faster when most of the files are related to the TLA.  Doubly so when the drive is not indexed.

All that being said, I'm not sure it's really your problem.  I think I need to change TopLevelAssemblyUtilities.GetLinks() to accept a list of folder names, rather than a single one.  Working with a list of folders shouldn't be too much trouble on 'top down'.  For 'bottom up' I'll need to figure out if Design Manager's WhereUsedCriteria will accept a list of folders, or if I have to iterate over them.

Thinking about it a little more, rather than a list, I would probably want a Dictionary.  The keys might be True or False to indicate if subdirectories are included, and the value for each would be a list of directories to search.

I'm not sure about the GitHub question.  I'll ask my son when he gets back later today.

rmcanany, 2022-12-11:
Well, he uses git all the time, but just for his own work.  He doesn't use branches.

How about we try it on ToyProject?  You could make a branch with some small change and create a pull request.  I would then try to fetch it without merging.  I'll check if I can toggle between the new branch and master, testing both.

farfilli, 2022-12-11:
I have created a new branch on ToyProject to experiment with how the pull request of a different branch works.
At the same time on the experimental housekeeper branch, I changed the code for TopLevel ASM which now search files on all the folders in the source list.

rmcanany, 2022-12-11:
OK I just saw it.  FYI, it will probably be tomorrow afternoon before I can play with it.  I'll let you know how it goes.

I can't wait to try out all your changes on Housekeeper!  Man you are FAST!

farfilli, 2022-12-12:
In the last commit, I moved the file type filters on the toolbar (the original controls are still there and need to be removed).
Also, I have added the ability to delete file sources by selecting and pressing the DEL\CANC key
DEL\CANC key works on files too but in this case, they will be moved to the "Excluded files" group. Excluded files are processed only if selected, if 0 files are selected they will not be processed.
Use of DEL\CANC key again to reset the group.

I may add a button on the toolbar to do the same as DEL\CANC

[image]

rmcanany, 2022-12-12:
That is looking so great!  I especially like the toolbar on top of the file list.  No more hunting all over the form to find the setting you need!

I like your idea on the TODO list.  Again, it keeps the user's focus in one place.  Brilliant!

It's not clear to me where a top-level assembly is identified.  Is it the .asm icon on the toolbar?  

I was able to retrieve and test your pull request on ToyProject without merging.  So when you're ready, send me a pull request for Housekeeper and I'll start playing with it.

farfilli, 2022-12-12:
Yes it's the asm icon on the toolbar. Toolbar buttons have tooltip activated.

I'm going to pull the experimental branch.
There is a lot of old update request to be removed in the code.

farfilli, 2022-12-12:
Pull request done, there is a conflict, can you handle it?

rmcanany, 2022-12-12:
[top_level_assy.zip](https://github.com/rmcanany/SolidEdgeHousekeeper/files/10211967/top_level_assy.zip)

Hey Francesco,

I was testing top-level assembly and some files showed up that don't belong.

I chose project_1_toplevelassembly.asm as the top level assembly.  I also used Add Folders and Subfolders to choose Project_1 and Common_Parts.  I used the Top Down search mode.

In the status bar, it reported 3 files were found, which is correct.  However, in the file listing it showed two Common parts that were not related to the top level assy.

I didn't try to troubleshoot, but I did zip up the files.  They're attached (I think).

farfilli, 2022-12-12:
Anyway I got the point, when there is a top level asm the folders should be used for that search only, tomorrow I will think at how to introduce this schema

farfilli, 2022-12-13:
Add the right mouse button menu on the file list.
It works on single and multiple selection

[image]

rmcanany, 2022-12-13:
That is looking so good!  

I saw your other note about dealing with conflicts and couldn't agree more.  I got my local copy so screwed up, I had to restore from backups.  Just by luck I had one from the day before yesterday.

Let me work on testing and cleanup for a while.  I'll let you know when I'm done.

farfilli, 2022-12-13:
It's time to think about what to do with the File search and Property filter options; I need to learn what they do and think about how to integrate better with the changes made till now

farfilli, 2022-12-13:
Ok I got it. About file search would be good to make it as a list filter? I will probably integrate it in a second toolbar on the bottom. In the same toolbar I will add the property filter.

Done that the list page will be very clean and neat

rmcanany, 2022-12-13:
I was thinking the exact same thing -- everything on the toolbar where they belong!

I'm going to move Font size to the Configuration Tab.  I see you left the original file type filters (*.asm, *.par, etc) check boxes in place.  Was there a reason for that, or can I remove them?

Once we get things pretty well settled, I need to look at saving and loading the configuration file defaults.txt.  That's how I persist control settings on the form.  I realize there is a Settings thing in Visual Studio, but I kinda like my way because I control what gets persisted and what doesn't.  It also makes form control values available to external programs (with a function I provide).  That one is on me, but I may have some questions about the ListView object.

rmcanany, 2022-12-13:
Oh one other thing.  Would it screw you up if I changed from your tmpFolderDialog to my FakeFolderBrowserDialog?  

The latter is more up to date, in my opinion.  In particular it has the 'Pined Folder' pane, which I use all the time.  Also, the top navigation pane lets you select, say, 3 levels up with one click and no scrolling.  It's also easy to display the last folder visited when it launches.  

To see the difference, click one of the Browse buttons on the Configuration Tab.

farfilli, 2022-12-14:
Original type filters are still there cause I didn't have the time to review the code where they were used, at the moment the filter on the toolbar just reflects their status. Will try to remove them soon.

I changed the FolderBrowseDialog to CommonOpenFileDialog, you can use this control that is a native one instead of creating a FakeFolderBrowserDialog. The "trick" is to set the property "isFolderPicker" to True.

farfilli, 2022-12-14:
old Type filters removed and controls moved to a new bottom toolbar

rmcanany, 2022-12-14:
You had earlier made several comments/suggestions to which I have yet to respond...

**Collection of results from each source**
I'll let you think about it.  The idea of creating a separate class sounds excellent.  There is way too much code in Form1.

**Get Top Level Assy links without Where Used**
I could easily do that in TLAU.  If you pass me an empty string for TopLevelFolder, I would simply skip the Where Used step.  

The user could possibly indicate such an outcome by specifying a Top Level Assy, but not adding any folders.

**Right mouse button menu to Open/Exclude/etc.**
I've already used it.  Handy!

**Truncating the path in TextBoxStatus**
I wonder if it might be better to show just the filename.  Does anyone care where it is?  I'll look at that.

farfilli, 2022-12-15:
**Collection of results from each source**
I will open an issue for this task

**Get Top Level Assy links without Where Used**
What if we add an option to activate\deactivate the WhereUsed function? In this way who want to read only the links from the ASM file could do it

**Right mouse button menu to Open/Exclude/etc.**
I like those little helpers, feel free to add more to the RMB menu

**Truncating the path in TextBoxStatus**
I wonder if it might be better to show just the filename. Does anyone care where it is? I'll look at that.

rmcanany, 2022-12-15:
Looks great.  Works great.  Testing underway.  Will open a separate Issue if tweaks or updates are needed.

farfilli, 2022-12-19:
A way to load direct links from the ASM file is needed, my files are spread in many folders with thousands of files so starting from the link contained in the ASM is much faster

rmcanany, 2022-12-19:
Reopening this Issue.

In the Readme, I said that is done by selecting the top level assembly, but without specifying any folders to search.  Do you think that's actually possible?

farfilli, 2022-12-19:
In line 56 I automatically add to the folder to search the TLA folder, if you remove this line then there will be no folder specified.
However, this actually doesn't work

[image]

Another proposal would be to add a specific button for this search
[image]

farfilli, 2022-12-20:
I have implemented my solution to find a link over a file.
I like the solution with the RMB because in this way a user can retrieve links to any selected file.

My solution is rude, and it retrieves links of assembly only, if you have a better solution just replace mine

rmcanany, 2022-12-20:
OK.  I'll play around with it.

I tried to move the File Type filters and their label from the upper toolbar to the lower, but it wouldn't let me.  Is there some trick I don't know?

farfilli, 2022-12-21:
File Types filter are buttons that belong to the upper toolbar, you can't move them from one toolbar to another, perhaps with a copy-paste or manually modify the form1.designer file.

Clarification:
The upper toolbar contains the tools that act on the generation of the list
In the lower toolbar instead, there are tools that act on the current list
So, as they work now, the file type filters should stay on the upper toolbar.

rmcanany, 2022-12-21:
Got it.  I'll update my instructions accordingly.

rmcanany, 2022-12-26:
Integrated the filters back into UpdateListViewFiles().  

Also came up with a way to have "bare" top level assemblies (where no top-level folders are selected) work with the Update button.

Also sped up FollowLinksBottomUp() 10X by not opening every file in DM.  (Doesn't speed up Where Used, unfortunately).

Testing now.

rmcanany, 2022-12-26:
This is looking awesome!  If you don't have anything more you want to add, I think I can close it.  Let me know.

farfilli, 2022-12-27:
Yup, close it

rmcanany, 2022-12-27:
OK, closing.  It turned out great!  Thank you!

closed 2022-12-27 by rmcanany