rmcanany/SolidEdgeHousekeeper #17: Move *.vb to the My Project directory

rmcanany, 2022-12-10:
When visiting the home page on GitHub, the file list is too long and no one sees the Readme.

rmcanany, 2022-12-26:
I moved everything but the forms.  I couldn't figure out how to move *.Designer.vb or *.resx without causing problems.

farfilli, 2022-12-30:
I have updated the icon of the property filter form and I just realized that the forms: ApplicationHelperFunctions, Defaults, FileHelperFunctions and LogFile are not used as a form but just as a class.

There is a reason for doing this? Perhaps you want to use the form at some point?
Otherwise it would be neat to change them into a proper class

rmcanany, 2022-12-30:
Yeah, there is a very good reason: I didn't know what I was doing!

Those are "partial classes" of Form1.  I wanted to organize the code a little bit, but couldn't figure out how to access the information back on the main form, e.g. TextBoxTemplateAssembly.Text.  

Now I know (more or less).  Creating proper classes is on my list of clean-ups, but I haven't gotten to it yet.

Incidentally, I finally figured how to move the forms to the My Project directory.  I'm going to close this Issue.  I'll make a new one about creating proper classes in a bit.

closed 2022-12-30 by rmcanany