rmcanany/SolidEdgeHousekeeper #22: Create a class to handle file sources

farfilli, 2022-12-15:
The class should have its properties and methods to be easily handled

rmcanany, 2022-12-16:
I'm ready to start on this, but want to wait until you've wrapped up work on ListViewFiles and friends.  I'm going to start by better understanding how it works, so you won't be holding me up if you're not quite ready.  Let me know.

farfilli, 2022-12-17:
This class should contain sources, results and methods to retrieve, update and filters.
In this way the Listview will be just the output to show user what is happening as it should be.

I can take this task if you want

rmcanany, 2022-12-17:
Well, OK, but don't feel obligated.  I'll assign it to you for now.  If you change your mind, it's no problem.  

In the meantime I'm going to get busy getting the Readme updated with pictures and descriptions.  I don't think it will be too hard on the Github side -- they use markdown which I'm pretty sure supports images.  Not sure on the VB side.  Do you think a RichTextBox might be the place to start?

farfilli, 2022-12-17:
I do usually create pdfs for instructions but if you want something more integrated perhaps an HTML page would be better?

rmcanany, 2022-12-17:
I've been playing around with the Readme on ToyProject.  Check it out!

farfilli, 2022-12-18:
It looks good!

rmcanany, 2022-12-22:
Hey Francesco,

Could you please hold off on this for a while?  I'm integrating the property and wildcard filters back into the ListView framework.  I don't want to create a bunch of merge conflicts.

farfilli, 2022-12-23:
Yep, it's not a priority at the moment to create this Class

rmcanany, 2024-01-03:
Target for v2024.2

rmcanany, 2024-08-14:
Closing as complete.  New class is UtilsFileList.vb.

farfilli, 2024-11-06:
One of the scopes of this issue was to separate the list of sources from the list of files. This seems not to have been done.
I would reopen it to complete the separation of the two lists.

At the same time, I would introduce a new ListViewSources (as listview) on the top of ListViewFiles.

Benefits:
- easier to handle the updates; there is no need to check for source vs file
- The new option to show-hide groups will not mix sources with the files

rmcanany, 2024-11-06:
You're right, I didn't get to it.  Reopening.

farfilli, 2024-11-08:
I worked a bit on that, the most on the UI but the basic functionalities work.
Being a bit invasive I created a [new Branch](https://github.com/farfilli/SolidEdgeHousekeeper/tree/ListViewSources), let me know if it's ok and I will make a PR to merge in master.

Looks neat BTW
[image]

rmcanany, 2024-11-26:
Closing as complete.