rmcanany/SolidEdgeHousekeeper #26: Update Readme to show and describe the new interface

rmcanany, 2022-12-17:


farfilli, 2022-12-19:
Robert, great work on updating the ReadMe; it just misses the description for these buttons:
- Add files from a list
- Export list
- Show only files with errors
- Delete all
- Update

rmcanany, 2022-12-19:
Thanks for that.  I'll get those added probably today.

I'm having a lot of trouble when I edit Form1.Designer.  About every other time I save my changes, Visual Studio renames those ListViewGroups like it did before.  It's driving me crazy.  Do you have any ideas on what I might try to fix it?

farfilli, 2022-12-19:
We could get rid of the "fake items" on the listview, I have added them just for preview.
And then create the 6 needed groups during the form load

It is important to create them in sequence like they are now

Files sources
Excluded files
Assemblies
Parts
Sheetmetals
Drafts

rmcanany, 2022-12-19:
OK.  It's worth a try.  I'm almost afraid to use the Form Designer at the moment.  Do you want me to try it on this end first?

farfilli, 2022-12-19:
Just remove everything and create groups at runtime

rmcanany, 2022-12-19:
OK.  I'll try it this afternoon.

rmcanany, 2022-12-19:
Before I start breaking things, let me summarize what I think 
I'm going to do:

In Form1.vb (Design), ListViewFiles object:

Remove all entries in the Groups and Items collections.

In Form1_Load():

Add this stuff:

        Dim ListViewGroup1 As System.Windows.Forms.ListViewGroup = New System.Windows.Forms.ListViewGroup("Files sources", System.Windows.Forms.HorizontalAlignment.Left)
        Dim ListViewGroup2 As System.Windows.Forms.ListViewGroup = New System.Windows.Forms.ListViewGroup("Excluded files", System.Windows.Forms.HorizontalAlignment.Left)
        Dim ListViewGroup3 As System.Windows.Forms.ListViewGroup = New System.Windows.Forms.ListViewGroup("Assemblies", System.Windows.Forms.HorizontalAlignment.Left)
        Dim ListViewGroup4 As System.Windows.Forms.ListViewGroup = New System.Windows.Forms.ListViewGroup("Parts", System.Windows.Forms.HorizontalAlignment.Left)
        Dim ListViewGroup5 As System.Windows.Forms.ListViewGroup = New System.Windows.Forms.ListViewGroup("Sheetmetals", System.Windows.Forms.HorizontalAlignment.Left)
        Dim ListViewGroup6 As System.Windows.Forms.ListViewGroup = New System.Windows.Forms.ListViewGroup("Drafts", System.Windows.Forms.HorizontalAlignment.Left)

Don't add:

        Dim ListViewItem1 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 1", "C:\Documents\Projects\FakeProject"}, "Unchecked")
        Dim ListViewItem2 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 2", "C:\Documents\Projects\FakeProject"}, "Unchecked")
        Dim ListViewItem3 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 3", "C:\Documents\Projects\FakeProject"}, "Unchecked")
        Dim ListViewItem4 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 4", "C:\Documents\Projects\FakeProject"}, "Unchecked")
        Dim ListViewItem5 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 5", "C:\Documents\Projects\FakeProject"}, "Checked")
        Dim ListViewItem6 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 6", "C:\Documents\Projects\FakeProject"}, "Error")
        Dim ListViewItem7 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 7", "C:\Documents\Projects\FakeProject\Drawings"}, "Checked")
        Dim ListViewItem8 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem(New String() {"FakeFile number 8", "C:\Documents\Projects\FakeProject\Drawings"}, "Checked")
        Dim ListViewItem9 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem("C:\Documents\Projects", "folder")
        Dim ListViewItem10 As System.Windows.Forms.ListViewItem = New System.Windows.Forms.ListViewItem("C:\Documents\Projects", "folders")

Do add:

        ListViewGroup1.Header = "Files sources"
        ListViewGroup1.Name = "Sources"
        ListViewGroup2.Header = "Excluded files"
        ListViewGroup2.Name = "Excluded"
        ListViewGroup3.Header = "Assemblies"
        ListViewGroup3.Name = ".asm"
        ListViewGroup4.Header = "Parts"
        ListViewGroup4.Name = ".par"
        ListViewGroup5.Header = "Sheetmetals"
        ListViewGroup5.Name = ".psm"
        ListViewGroup6.Header = "Drafts"
        ListViewGroup6.Name = ".dft"
        Me.ListViewFiles.Groups.AddRange(New System.Windows.Forms.ListViewGroup() {ListViewGroup1, ListViewGroup2, ListViewGroup3, ListViewGroup4, ListViewGroup5, ListViewGroup6})

Don't add:

        ListViewItem1.Group = ListViewGroup3
        ListViewItem2.Group = ListViewGroup4
        ListViewItem3.Group = ListViewGroup4
        ListViewItem4.Group = ListViewGroup4
        ListViewItem5.Group = ListViewGroup5
        ListViewItem6.Group = ListViewGroup5
        ListViewItem7.Group = ListViewGroup6
        ListViewItem8.Group = ListViewGroup6
        ListViewItem9.Group = ListViewGroup1
        ListViewItem10.Group = ListViewGroup1
        Me.ListViewFiles.Items.AddRange(New System.Windows.Forms.ListViewItem() {ListViewItem1, ListViewItem2, ListViewItem3, ListViewItem4, ListViewItem5, ListViewItem6, ListViewItem7, ListViewItem8, ListViewItem9, ListViewItem10})

farfilli, 2022-12-20:
I have removed the preview items that causes problems during pull request

rmcanany, 2022-12-20:
It's working so far.  Much less scary to tweak things!

I've been updating the Readme.  When you get a chance, please check the File Selection section.  I'm sure there are things to add/remove/fix/clarify.

farfilli, 2022-12-22:
I saw the new Icon, and it looks good; in my opinion, it needs a few tweaks:
- "handle" color should be a bit more bright, it almost disappear on my blackish windows toolbar
- "handle" should be a bit longer to be visible
[image]

rmcanany, 2022-12-26:
The Readme is getting closer.  If you have a chance, please look it over.  Especially the File Selection and Filtering part.

farfilli, 2022-12-27:
It looks good to me, just miss my latest addition on RMB menu item to find ASM links

rmcanany, 2022-12-27:
Good.  I've been looking at it for so long, I can't tell up from down.

I don't think we need the RMB button for ASM links any more.  I integrated
that functionality into the Update Button code.  I didn't delete anything,
just added a MsgBox to BT_FindLinks handler.

I'm pretty sure it works the same, but if you get a chance, please try it
out.  I feel it's more consistent with your overall design direction --
keep the user's eyes on the Selection Toolbar.

farfilli, 2022-12-30:
I would leave the RMB find links there, it could be usefull in some situations and its pretty self explanatory.
But hey, you are the master branch :)

rmcanany, 2022-12-30:
OK, let's give it a try.  I'll remove my MsgBox and update the Readme.  If it causes any confusion, I can just blame you :)

rmcanany, 2022-12-30:
Done for now.  Closing.

closed 2022-12-30 by rmcanany