rmcanany/SolidEdgeHousekeeper #31: Update Release Notes

rmcanany, 2022-12-28:
Getting closer...

@farfilli I went through the commit history to try to remember everything. If you have time, please take a look at the [**Release Notes**](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/release_notes.md) and let me know if you notice anything I messed up/left out.

rmcanany, 2022-12-29:
@farfilli 

To make it less confusing for people who download the previous release, I changed the names of `README.md` and `release_notes.md` to `README_NEW.md` and `release_notes_NEW.md`.  I also added `README.md` and `release_notes.md` from v0.1.10.3.

I'll undo the changes when I get the release candidate ready.

In the meantime, if you want to see my recent changes, please look at `*_NEW.md`.

farfilli, 2022-12-30:
They looks good to me

rmcanany, 2022-12-30:
Good deal.  FYI, I'll be out of the office next week.  To be responsive to questions/comments I'm going to wait until the following week to post to the SE Forum.  I'm still working on the text of that post.  Below is what I have so far.  I might take out the Question part and save it for later.  

==================================

Solid Edge Housekeeper: Some Big News, a Call for Beta Testers, and a Question

Big News
Housekeeper has its first GitHub Contributor -- our very own @Fiorini!  I can't wait for you to see what he has done with the place!

Beta Testers
To be among the first to see the new release in action, volunteer to be a Beta tester!  Beta testing is nothing more than performing your own workflow on your own files and telling me if you run into problems.  It isn't meant to be a lot of work.  The big idea is to make the program better for you and me and everyone else!  To sign up, message me or respond to this post.

A Question
For a future release, I'm looking at how we find files related to a top-level assembly on a server.  The way it works now, the program finds all internal links to the assembly, which is fast.  After that, the only thing left is to find the drawings.

For that, Housekeeper uses *where used*, which is slow, at least on non-indexed drives.  The thing is, there is an implicit assumption that we don't know the drawing file names or where they are located.  

At my company we know exactly.  The draft has the same name as the model and is in the same directory.  We don't need *where used* for that.

Here's the question (even if you don't use Housekeeper): How about you?  If you had the name of a model file, would you know the drawing's name and where it is located?  What about revisions, or other complexities?

Please respond to this post with your company's method.  The hope is to come up with a file name "formula" and eliminate the need for *where used* altogether.  

If someone has already posted the method you use, to minimize work for you, please feel free to simply "Like" their response.  Or maybe respond and say "we do it like @SoAndSo, except this tiny difference".

Looking forward to your input!

rmcanany, 2022-12-30:
Done for now...

closed 2022-12-30 by rmcanany