rmcanany/SolidEdgeHousekeeper #37: House keeper wont pull drawings

nathanarinta, 2023-02-07:
I'm not a programmer or anything but I'm trying to get housekeeper to pull all the drawings so I can auto save to pdf and print them. This would literally save me days of work. Im not sure if im supposed to be putting my issues here it very well could be something i am doing work. Thank you for any help!

rmcanany, 2023-02-07:
He @nathanarinta 

It's fine to ask questions here.  You could also use the Solid Edge forum.  A lot of folks on that site use Housekeeper and are always willing to help.

It should definitely be able to save drawings to pdf and print them.  That's one of its most popular features.

Could you explain in more detail the behavior you saw?  Did you get an error message?  If possible, it might help if you could you upload a screenshot of the following tabs: Home, Draft, and Configuration.

nathanarinta, 2023-02-07:
No error message or anything. I use the top level assembly search and grab my main assembly. Then find all the linked files. Housekeeper seems to find all the assembly, parts, and sheet metal parts but no draft files. I am very new to house keeper so i very well could be missing a step. i have been reading through forum post and the readme documentation but was unable to find anything.
[image]
[image]
[image]

rmcanany, 2023-02-07:
@nathanarinta  OK, a couple of things...

If you right-click the top-level assembly and select "Find linked files", it only follows links within the top assembly and any subassemblies.  It does not search for Draft files.  This isn't exactly a bug, but I admit it is confusing.  To get the Draft files, try clicking the "Update" button instead.

For Save As, you have to check either "Original directory" or Browse to a directory where you want to save them.  I can see I need to add a check for this.  It's on the list!

You mentioned printing, too.  Please be sure to read the help on that command.  There are some limitations that can be confusing.  The help information is in the README, almost at the very end.

If you can't get it to work, please post back.  If you're having trouble, many others probably are, too.  If, on the other hand, it does work, let me know and I can close this issue.  Thank you!

nathanarinta, 2023-02-07:
Thank you for the fast response! When I hit the update button it still doesn't find any draft files. It loads for a few seconds and says 1 files found at the bottom. Could it be the way that the files are stored in the server?
[image]

rmcanany, 2023-02-07:
Hmm.  For a test, try to clear the list (the trash icon), then reselect the top level assembly and Update.

nathanarinta, 2023-02-07:
Sadly no luck. It does the same thing. It maybe looks like it opens design manager for a second then it closes it down.

rmcanany, 2023-02-07:
Well, not sure but here are a couple more things to try.

The search is isolated to the top-level folder and below.  Are the draft files below the top-level folder?  If not, you would need to add a folder under which the draft files can be found (it's the button to the right of the top-level assy).

I've seen situations where Design Manager hangs, but is in the background so you can't see to close it.  You could take a look in Task Manager (it could be listed as a "Background Process")  Alternatively, a reboot would close it.

rmcanany, 2023-02-07:
Yeah, so you would have to include the "test" folder for it to find the part files.  There are two ways to search, neither of which is perfect.  You can see the details here:
https://github.com/rmcanany/SolidEdgeHousekeeper#2-select-by-top-level-assembly

For your situation, where the part and draft files are in the same directory with the same name, I have a fix coming the next release.  Unfortunately, that won't be ready for a few months.

nathanarinta, 2023-02-07:
After rereading what you said a few times I figured what you met and I got it to work correctly! Thank you so much for helping me though this! This program is such a god sent!

rmcanany, 2023-02-07:
Hey, great!  

I have a request -- how about signing up as a Beta tester?  Once I get that fix I mentioned, it would be great to have someone who could try it out before it's released.  It won't hurt my feelings if you can't, but it would be awesome if you could!

rmcanany, 2023-02-08:
Resolved.  Closing.  

For details, see: https://community.sw.siemens.com/s/question/0D54O00007qDzoeSAC/solid-edge-housekeeper-big-news-a-call-for-beta-testers-and-a-question

closed 2023-02-08 by rmcanany