rmcanany/SolidEdgeHousekeeper #38: PMI and sketch no hiding

tempod, 2023-03-20:
HI.
First of all congratulations and thanks for this fantastic program.
I have noticed some strange behavior. Even selecting the options to hide the buildings they are still shown.
I made a little video for you to see. It seems that they are correctly hidden but then on the saved file they are still visible
[image]

rmcanany, 2023-03-20:
Thank you for the feedback.  You're right.  I've noticed that, too.  I'll add it to the list!

I don't use Sync.  When working interactively, do you know the proper steps to hide those elements?

tempod, 2023-03-20:
I don't use synchronous mode either.
to hide everything right click on "Design Bodies" and then on "Show Only"
[image]

rmcanany, 2023-03-20:
OK, cool  I'll try that.

rmcanany, 2023-04-16:
Hi @tempod ,

I just pushed a release that includes a fix for this issue.  If you have a chance, please test it out and let me know how it goes.  Thank you!

tempod, 2023-04-18:
Hello.
I tried and the problem disappeared!
Works great!!
Congratulations again!

closed 2023-04-19 by tempod