rmcanany/SolidEdgeHousekeeper #40: Dead end, if add files by d&d and "Unsorted"

TomsHelfi, 2023-06-13:
If  I set the order option to "Unsorted" and select files by d&d and then process, the warning appears,  that I have to update the list:
[image]
After updating the list, the files disapears.

If I do it the same way in 2023.3, the process start SE, without warning

rmcanany, 2023-06-13:
Thanks for that.  I'm seeing the same thing.  I'll dig into it.

rmcanany, 2023-06-14:
Release v2023.4.1
Ready for testing:  Drag-and-drop files getting deleted when running the Update File List command.

Release notes
https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/release_notes.md

Zip file
https://github.com/rmcanany/SolidEdgeHousekeeper/releases/tag/v2023.4.1

TomsHelfi, 2023-06-15:
Works as expected! Thank you