rmcanany/SolidEdgeHousekeeper #41: Error proccessing ASM file

TomsHelfi, 2023-06-20:
An error occurs, if the assembly 177325.001-V01.asm is processd by SolidEdgeHousekeeper 2023.4.1. The file was added by drag&drop. The germans sentence means "The object reference was not set to an object instance" and "End of internal exception stack monitoring"

- Error: [Housekeeper_20230620_124737.log](https://github.com/rmcanany/SolidEdgeHousekeeper/files/11799086/Housekeeper_20230620_124737.log)
- Assembly: [177325.001-V01.zip](https://github.com/rmcanany/SolidEdgeHousekeeper/files/11799077/177325.001-V01.zip)
- Templates:  [Template-Robatech.zip](https://github.com/rmcanany/SolidEdgeHousekeeper/files/11799106/Template-Robatech.zip)
- Settings: [defaults.txt](https://github.com/rmcanany/SolidEdgeHousekeeper/files/11799120/defaults.txt)

rmcanany, 2023-06-20:
Thanks for that.  Looks like you are on SE 2023.  I'm still on 2022.  I could open all the files except 177325.001-V01.asm and the assy template.

I ran 176857.002.asm, but with my assy template.  Worked fine.  Would you have a template version prior to 2023 that I could try?

TomsHelfi, 2023-06-21:
Yes, I have of Version 2021. See attached [Template-Robatech_SE2021.zip](https://github.com/rmcanany/SolidEdgeHousekeeper/files/11813802/Template-Robatech_SE2021.zip)

TomsHelfi, 2023-06-26:
The issue dissapears, if I change the user interface language of SE to english. Looks like, the german UI language is the problem. I procced Housekeeper also with german UI in VS. It runs in to this exception: 
[image]

TomsHelfi, 2023-06-27:
I tried it. Now, it runs without exception in the german UI!

rmcanany, 2023-06-27:
OK, thank you.  Closing for now...

closed 2023-06-27 by rmcanany

TomsHelfi, 2023-06-29:
Question: can you say me, when you release the new version, where this issue is fixed?

rmcanany, 2023-06-29:
The fix is to put SEDoc.AsmRefPlanes in a Try/Catch block.  The code is in 
AssemblyTasks.HideConstructionsInternal and AssemblyTasks.FitPictorialView.