rmcanany/SolidEdgeHousekeeper #42: Process Tasks in background

farfilli, 2023-07-27:
This option just make Solid Edge not visible but the processing time is not speeded up.

I would introduce the real open in background to have a fast mode.
I did a test with 27 asm file to change a property this is the results:

- Solid Edge visible: 114 seconds
- Solid Edge hidden: 102 seconds
- Real open in background: 42 seconds

The open in background probably have some limitations with tasks that requires graphics so all the tasks must be tested.

I created a branch for experiment, please have a look at it

Please note that the code now always opens documents in background; do not use the option process tasks in background to test it

rmcanany, 2023-07-29:
OK.  That looks very promising!  What do you think, just try a bunch of files on all tasks and see what bombs out?

farfilli, 2023-07-30:
Yes exactly, I do not have the time to do it. Perhaps you can make this open in background available only for the tasks that works.

And now a question, if you activate multiple tasks on the sime file does it open and close multiple times?

rmcanany, 2023-07-30:
No problem.  I'll test it out.

It only opens the file once.  Before it does, I run multiple checks in CheckStartConditions().  Background-processing compatibility is already in there.

rmcanany, 2023-08-02:
I attempted to integrate your code into mine.  It's in my latest commit.  

In Form1.vb, I updated ProcessFile().  In Form1.ApplicationHelperFunctions.vb, I updated SEStart().  Both changes primarily incorporate the CheckBoxBackgroundProcessing flag.

I probably messed something up.  On a quick test, it worked on *.par, *.psm, *.dft, but not for any *.asm task I tried.  SE bombed out every time.  To make sure it wasn't something about my own files, I tried Grinder.asm in the SE Training folder.  Same result.

Ideas?

farfilli, 2023-08-03:
I have done a quick test and just can confirm that something is wrong.
However, I can confirm that the technic works for asm too as I have used in my program Assembly Colorizer, you can have a look at the code here: [https://github.com/farfilli/Assembly-Colorizer-for-Solid-Edge/blob/master/ASM_Colorizer.vb](url)

rmcanany, 2023-08-03:
Correction: It only crashes on assy tasks that do a Save.

I undid my changes and copied Form1.ProcessFile() from your Experimental branch.  Still getting the crashes.

I ran across a couple of things from jnewell on the topic:
https://community.sw.siemens.com/s/question/0D54O00006viUVaSAM/open-documents-in-backgroundsilently

https://blogs.sw.siemens.com/solidedge/how-to-open-documents-silently/

For now, I have changed it so that assembly files run in the "old" background mode -- graphics hidden, not disabled.  It's not ideal, but stopped the crashes.

Unless you want to play with it some more, I think I can close this for now.  Let me know.

Oh, I had a branding idea for any project to which you contribute :)

[image]

farfilli, 2023-08-04:
Nice logo! 🤣

I have tested only the property change with assemblies and it works. It also works in my other repository, perhaps something that interferes is set in the code. It may need a check of all that is done on the SolidEdge.Application from when the document is opened till is closed.

I did read the posts of Jason, it is interesting to know how it works; basically, we can just pass the second argument as an 8 of the open method to achieve what community extension does

rmcanany, 2023-08-28:
Complete in v2023.6.  Closing.