rmcanany/SolidEdgeHousekeeper #43: Option to create missing properties

farfilli, 2023-08-28:
What about an option to create a Property if doesn't exist?

rmcanany, 2023-08-28:
That's funny you mentioned that.  I've been looking how to add Quantity to drawings.  I'm not in love with the idea, but have had several requests, including from my own guys.  Anyway, I realized it would be handy to add the Property if it didn't exist.  I was going to add it to the Property Find/Replace functionality.  Would that cover what you were suggesting?

farfilli, 2023-08-28:
I don't like the idea of adding quantities on the drawing as well; it is not where that information should be.
Yes the Find/Replace would become Find/Replace or Add

rmcanany, 2023-11-29:
@farfilli This is done I think.  If you get a chance to try it out, let me know how it goes.  

I added an option on the Configuration Tab, Miscellaneous Group, to enable/disable it.  I updated the readme also.  It's the last paragraph in this section: https://github.com/rmcanany/SolidEdgeHousekeeper#property-find-replace

rmcanany, 2023-12-04:
Done

closed 2023-12-04 by rmcanany