rmcanany/SolidEdgeHousekeeper #68: Consolidate Tasks UI

rmcanany, 2023-12-23:
Place task controls and settings into a hierarchical structure.  Top level is a task category.  For example, maybe Edit/update, Check, and Wrap up.  Add images to the category and task lines for quick identification and a splash of color.

Add check boxes for file type, only where the task is applicable to that file type.  If a task is selected, drop down its settings below the task name.  The Windows form control TableLayoutPanel has this ability.  [Basic tutorial](https://www.codeproject.com/Tips/842418/Designing-the-Layout-of-Windows-Forms-using-a).

Provide an edit button to present a form for more complex settings.  Return a Dictionary, which can be serialized to json for compatibility with the current Configuration scheme.

[image]

rmcanany, 2023-12-28:
Those are all great ideas!  I'll play around with it and keep you posted.

rmcanany, 2024-01-03:
Target for v2024.2

rmcanany, 2024-01-13:
TableLayoutPanel examples

https://www.c-sharpcorner.com/UploadFile/mahesh/tablelayoutpanel-in-C-Sharp/

rmcanany, 2024-01-21:
@farfilli 
I think I finally know a way to support multiple instances of a single command in a task list.  

Currently, the tasks are held in a Dictionary(Of String, L2A).  L2A is a LabelToAction object.  It has that name for a reason that didn't actually work.

What I think I need is to have a Task object with some basic properties.  Then I need to sub-classs that to make SaveAs, PropertyFindReplace, etc. Task objects.  Each of the sub-class objects would hold all relevant information, such as SaveAs.FileType, SaveAs.DirectoryFormula, etc.

Right now I store that information in various places on the UI.  That's not the way to do it.

Does that sound feasible?  What am I overlooking?

rmcanany, 2024-01-21:
@farfilli 
I added some functionality to the header row.  Now if you click the "+" button, it expands all task options.  Click again to collapse.

Click the task-select header button and it clears the selections.  I didn't think anyone would ever want to select all tasks, but could add it easily.

Click a file type button and it selects that file type for all selected tasks.  Click again to clear.

Click the help button and it brings up the Readme page on GitHub.  Click an individual task help button and it brings up the Readme of that command.  Oh, any idea why the help icon is messed up?  The others look good.  I create them all the same way:
`                    Button.Image = Form1.ImageList1.Images("icons8_help_16.png")
`

Here's a picture.  (This is from my Scratch code, not Housekeeper.  It's going to break a few things.  I wasn't ready to commit to that yet.)

[image]

rmcanany, 2024-01-24:
My attempt to add some "Francesco Flair."  What do you think?

[image]

farfilli, 2024-01-24:
It looks great!
I would really have more time to help you. Place all that controls and manage them one by one is a nightmare. My idea is to create a configurable custom control for actions. Then you add it and configure for each action in the table.
I will try to make a demo project for you.

rmcanany, 2024-01-24:
That would be great!  I'm having trouble wrapping my head around that part!

rmcanany, 2024-01-24:
I created a new repo, Scratch, in case it would help you to see what I have so far.

farfilli, 2024-01-25:
Robert have a look at this repository I just created; it is a row and dirty just to make you understand how it works.
There is space for a lot of improvement according on how you want to proceed; we can discuss about it

https://github.com/farfilli/TaskListDemo

rmcanany, 2024-01-25:
Thank you very much!  I'll study it today.  If there is something I don't
understand, I'll get back to you.

farfilli, 2024-01-26:
I have added the combo box and configuration controls, polished the UI a bit

farfilli, 2024-01-26:
I have added a property to control vertical border visibility
[image]
[image]

rmcanany, 2024-01-26:
Nice!

farfilli, 2024-01-29:
Robert I have moved the Task types inside the TaskControl class, it is where it should be.

Let me know if you need modifications on the UI, I'm thinking that moving the combo beside the name would look better.
Perhaps keep the combo line and add another one for the name line as optional.

rmcanany, 2024-01-29:
Makes sense.  I'll check it out!

FYI, I'm refactoring the whole Task architecture.  It's way cleaner.  I now have a base class, Task.  Each task will now inherit from that and have all the options and code contained within.  You know, like it should have been from the beginning!

Nothing is on GitHub yet.  I still have some integration work to do to.

rmcanany, 2024-01-29:
That's my current thinking -- each task subclass contains its own TaskControl.  I haven't gotten that far yet, but it sounds like we're on the same page.

farfilli, 2024-01-29:
I would suggest to have an unique configurable TaskControl class that is used in each taskClass

rmcanany, 2024-01-29:
Yeah, I had the same idea.  I think I can do the common stuff in the base class, then handle the task-specific options in the subclass.

rmcanany, 2024-03-11:
Got it working.  Take a look if you get a chance!  I don't have the list customization part done yet.  Now that the Tasks are organized like they should be, it doesn't look too difficult.

We're releasing a big project at work this month -- perfect real-world test!

farfilli, 2024-03-12:
I had a look; I wouldn't say I like the idea of creating the tasks UI all by code; it is difficult to customize as you want. Anyway, it looks promising. I did try to modify the tasks page "expand all" behavior as each task is repainted in sequence and makes an odd display during it. I wanted to add a suspend table layout panel before paint and reactivate it after but I didn't find where the event fires.

rmcanany, 2024-03-12:
Yeah, the "expand all" is funky.  I tried Form1.SuspendLayout, but it didn't help.  I didn't think to do that on the TLP.  I'll give that a try.  I hid all the event handlers in a separate class called Task_EventHandler, by the way.  Seemed like a good idea at the time.

What I like best about it is how all options for a given task are right under your fingertips.  That, and the consolidation from four tabs to one.  There's a lot less jumping around the program to get things set up.

farfilli, 2024-03-12:
Just to refine the UI, I would add a transparent background on checkboxes rather than a solid grey control color. The best would be to have a highlighted icon on the mouse hover

rmcanany, 2024-03-12:
Thanks for the suggestion.  Keep 'em coming!

farfilli, 2024-03-13:
I had a second look at the code; and did some tests to speed up the expand\collapse.
I think the core of the slow performance is due to the amount of controls in the TLP.
I would strongly recommend creating a custom control that handles each task. This will reduce dramatically the number of controls in the TLP, the custom control will be treated as a single control even if contains many.
Side effect, you will have a visual designer to adjust the layout of the task UI

farfilli, 2024-03-13:
I created a Pull request with changes that makes the expand\collapse an acceptable experience; have a look at it

farfilli, 2024-03-13:
I made the https://github.com/farfilli/TaskListDemo to show you how the custom control works; have a look and if you need help I will integrate it in the main project for you to use

farfilli, 2024-03-13:
Be sure to update to the latest commits :-)

ChrisNeCle, 2024-05-14:
May I recommend making disabled vs unchecked boxes in the tasks tab more stylistically unique? It may be partially because of my colorblindness, but at certain angles on my monitor they're difficult to tell apart.

rmcanany, 2024-05-14:
That's a good idea.  It's on the list!

farfilli, 2024-06-20:
@rmcanany I can take the request to implement a custom control for task UI. You can create a new issue if you think it's needed to better track the task project

rmcanany, 2024-06-20:
No need.  It's already done.  I'm now cleaning out all the garbage I added previously.

rmcanany, 2024-06-23:
Just committed the code with the new `TaskControl` user control.  If you get a chance, please try it out.

@ChrisClems I tried to improve the checkboxes, but I'm not positive I understood the issue.  Would appreciate any feedback.

farfilli, 2024-06-24:
I did a quick test, it looks promising. I also edited the TaskControl a bit to have a less heavy UI.
Just some thoughts: instead of a panel I would use a list control; in this way, it would be easy to add\remove\reoder the tasks in the list. Also it will be easy to create a save\load of what I call "Plans"
(A plan is just a list of pre-configured tasks)

rmcanany, 2024-07-06:
Got the Edit Task List working (very minimally tested).  Would welcome any feedback if you get a chance to play with it.
[image]

rmcanany, 2024-07-06:
@farfilli I like your idea about "Plans".  I didn't notice that part of your comment before.  I've been thinking along the same lines.  I also agree it would be best to target for a future release.

farfilli, 2024-07-09:
@rmcanany I had a deeper look at the UCTaskControl implementation. The UCTaskControl should handle every aspect of the control, instead, I see that the TLPOption panel is generated at runtime by every single task class. There are hundreds of repetitive code lines that are not ideal to maintain and make it difficult to implement new tasks in the future.

UCTaskControl should contain all the options in a customizable way, if you feel that to limit the options more custom controls could be created to handle every single option line and to be added in sequence as needed.

In the end, you assign the task to the UCTaskControl and the control configures itself; also during execution, it can easily return the task and its configuration; in the same way it can return its configuration for saving purposes

I know it's a big change, and probably the actual state of the code is due to the past implementation, but I think a code refractory is needed; perhaps for a future release. Main object code reduction\consolidation you can set it as the first project?

rmcanany, 2024-07-09:
I appreciate the feedback.  

You're right, each Task that has options creates its own TLPOption panel.  I did it that way because I didn't see much commonality between them.  For example, SaveDrawingAs has lots of options, many of which are shown/hidden based on other options.  Other Tasks may only have a couple of CheckBoxes.

Do you think it would maybe be better to make a separate Option UserControl for each Task that has options?  I'm picturing something like UCSaveDrawingAsOptions that would be added to its UCTaskControl at runtime.  I guess the event handlers would be in there, too.  It wouldn't be that much work and there's no deadline.

Let me know your thoughts.

farfilli, 2024-07-12:
The tasks list should be reloaded after the edit without the need to restart SEHK.
This will also be needed when "plans" are implemented, that's mandatory to move from one plan to another without the need to restart SEHK and lose the file list and previous processing result.

rmcanany, 2024-07-15:
@farfilli I'm going to look at this next.  It shouldn't be too hard to repopulate the list live.

Regarding my other question about custom controls, would a UCTlpOptions for each task with options be a workable approach in your experience?  I recently had to modify some options for a command and, you're right, even though I wrote the original code only recently, it's not easy at all.

farfilli, 2024-07-15:
I see two possible approaches here:

1 the UCTaskControl contains all the possible options you would add. The auto configure method will hide the unused ones. I have used this approach in the User control demo project. Have a deeper look on how it's implemented.

2 create more User Control that rappresents an option row of the task. You need as many as the king of options you want to present, could be a simple checkbox with a label or a more complex like a button with a label and a drop-down list. Again the auto configure method will add rows of the correct kind accordingly with the task options.

Let me know if that make senso to you or if you need further explanation or an extended demo project in case you choose number 2

rmcanany, 2024-07-15:
Got it.  Appreciate the advice.  I'll study that example more closely.

In the mean time, I got the task list updating live.  Take a look!

farfilli, 2024-07-17:
It seems to work great!

rmcanany, 2024-07-28:
Closing as complete.  Opened a new issue for refactoring task options creation.

closed 2024-07-28 by rmcanany