rmcanany/SolidEdgeHousekeeper #75: Copy and Set Styles

tempod, 2024-03-13:
HI. I have two requests that are related to each other.

When I use "update face and view styles from template" only the currently used style is copied. Would it be possible to add an option to copy all the styles present in the template?
The second request concerns the possibility of setting a style when saving images of the pieces.
For example, when modeling the parts I use a gray background while when saving the images I set a style with a white background.
If it were possible to copy all the styles and set a style when saving it would be a nice progress in my opinion.

Thanks anyway for the excellent work.

rmcanany, 2024-03-13:
Those are great ideas.  They're on the list!  Thank you!

rmcanany, 2024-03-13:
@farfilli couldn't I just adopt your technique from UpdateDrawingStylesFromTemplate to tackle this one?

farfilli, 2024-03-13:
I think there is an API to easily manage the styles, I will have a look and return it

rmcanany, 2024-07-15:
@tempod I appreciate your great suggestions!  It should be working now.  If you can't compile from source, let me know and I'll zip up a pre-release for you.  

@farfilli I copied your technique from UpdateDrawingStylesFromTemplate.  Much easier and better.  I didn't even have to _touch_ your CopyProperties code.  Nice!

rmcanany, 2024-07-24:
Closing as complete.

closed 2024-07-24 by rmcanany

tempod, 2024-07-31:
wow! thank you so much. unfortunately i dont know how to compile from source...

rmcanany, 2024-07-31:
OK, no problem.  I've got one last issue to address, then am going to release it for beta test.  It should be ready in a week or two.  I'll let you know when it has posted and you can try it out then.

rmcanany, 2024-09-13:
@tempod 

I published the latest version of Housekeeper this morning.  Here are a couple of links to get you started:

[Release Notes](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/release_notes.md)
[Installation Instructions](https://github.com/rmcanany/SolidEdgeHousekeeper#installation)
[Release Page](https://github.com/rmcanany/SolidEdgeHousekeeper/releases)

If you get a chance to try it out, I would welcome any feedback you may have.

Robert