rmcanany/SolidEdgeHousekeeper #76: Add this tools create\copy\move\delete to edit properties task

farfilli, 2024-03-15:
Expand the edit properties capabilities to be able to:
- Add a property (already exists but works only with a value?)
- Copy the value of a property to another property (create it if it not exists)
- Move the value of a property to another property (create it if not exists)
- Delete a property

Side note: with the implementation of task lists where the same task can be done more than once, the interface can be simplified to process only one property per task. In the need to process more properties just add the edit properties task many times as needed

rmcanany, 2024-03-16:
Except for deleting, which I hope to add, that's actually possible now.  Here's an example of something I did the other day.  It toggles the material to GOLD and back.  I used it as a workaround for `UpdatePhysicalProperties` to get SE to create a Mass variable (instead of mass_1, mass_2, ...) for some older files.

Regarding the delete option, I'm kinda struggling with a clean clear way to incorporate it into the form layout.  If you have any ideas, I'm all ears.

[image]

[image]

farfilli, 2024-03-18:
Thanks, Robert, I didn't read the help carefully and forgot about this functionality; I would add a reminder beside the search legend about it

farfilli, 2024-03-20:
What about a text command to delete the property? Something like **%{DeleteProperty}** ?

closed 2024-03-21 by farfilli