rmcanany/SolidEdgeHousekeeper #80: Check for Solid Edge file compatibility

farfilli, 2024-03-20:
A check of file compatibility with the current Solid Edge installed version should be performed.
For some reason some student files are on my drive and if I try to process them by HK and a commercial license I get unexpected behaviour.

ChrisNeCle, 2024-05-13:
Do you have an example of a student file to test with?

rmcanany, 2024-05-13:
I don't.  If you have any, I'd be glad to do some testing.  Not right away, though.  Someone at work found out I wasn't too busy and, uh, fixed that.

rmcanany, 2024-07-11:
@farfilli  Is the strange behavior that SE closes and reopens on each file?  That's by design if the program gets a COM error.  That's been in there forever because I thought it meant SE was broken somehow.  I now doubt that's the case and I can probably disable that behavior.  What do you think?

farfilli, 2024-07-12:
[803110053X_.zip](https://github.com/user-attachments/files/16189434/803110053X_.zip)

This is an example file, it may be an SE2025 release or a Stundent Version, in any case not compatible with SE 2024 Commercial license

farfilli, 2024-07-12:
It would be good to at least handle the license version compatibility and exclude it from the SEHK restart. Anyway, I just tested and it doesn't restart with this file.

Please note, that structured storage edit, is not affected by the license version, so all tasks that can direct manipulate files doesn't need the version check

rmcanany, 2024-07-12:
I found a LicenseType property on the Application object.  I'm installing the Community Edition on an old machine and will see if that property or some other will give me some clue on the application side.

Do you think there might be a way to use structured storage to find out something relevant on the document side?

farfilli, 2024-07-12:
You can just swap license file to activate student Vs commercial licenses.

I don't think with structured storage that info to be available.

Last saved version instead is available

rmcanany, 2024-07-12:
Sure enough.  My LicenseType is Classic, the new install is Community Edition.

rmcanany, 2024-07-12:
OK, I'll try that.

rmcanany, 2024-07-12:
Not having any luck with the document side of things.  Not finding anything about commercial vs community vs academic, for one.  

Not even finding anything about the version.  I did find a property "SE Version" using Shell32.Folder.GetDetailsOf().  In the couple of files I checked, however, it was not populated.  The LastSavedVersion is a property of the SolidEdgeDocument object.  I could be wrong, but I believe that is only available after the document is open.

Taking a different approach, I tried removing the SEStop() SEStart() sequence after a file open exception, but the Application becomes brain dead after that.

rmcanany, 2024-07-15:
Closing as not currently feasible.  Can reopen if any other approaches come to light.

closed 2024-07-15 by rmcanany