rmcanany/SolidEdgeHousekeeper #82: CVS Formatting

Careframe, 2024-04-03:
Is juste a little suggestion but can it be possible to add CSV formatting of the output report text, like CSV to import this as a task list in excel ?

rmcanany, 2024-04-03:
Thank you for the suggestion!  That doesn't sound too hard.  I'll look into it.  

It won't be right away, but when the time comes if you would be willing to test it before release, I would appreciate any feedback.

rmcanany, 2024-07-06:
@Careframe We're going to be making some improvements to error logging, probably next year.  If you still need this, it would be great if you could provide an example of the format you're looking for.

rmcanany, 2024-07-15:
Superseeded with https://github.com/rmcanany/SolidEdgeHousekeeper/issues/113, closing.