rmcanany/SolidEdgeHousekeeper #83: Feature Idea: Rename Files

pkoevesdi, 2024-05-03:
(or is it already possible?)
I'm looking for a tool that runs through my folder and renames all files to their `title` property, of course also updating the references in asm and dft. Double filenames could automatically get a number added.

Background: After an export from a PDM system, the file names are only numbers from inside the PDM system. Which is awkward if passing these files to external people.

pkoevesdi, 2024-05-03:
It's very similar to the function "Part number does not match file name", but instead of only reporting it, it would be nice to extend this function to also fix this.

rmcanany, 2024-05-03:
Thank you for the suggestion!  I had a project not too long ago that could have used that functionality.  Like you, I thought it would be a nice addition to Housekeeper.  My working title for the command was "Fix this mess."

Until I get to it (no promises!), there's a workaround.  In Design Manager, you can copy/paste to and from Excel.  In Excel, use the Text To Columns and Concatenate functions to build the new name, then copy it back.  Here's a picture of how I did it:

[image]

pkoevesdi, 2024-05-03:
Ah, thanks for the hint. In my case, doesn't even need Excel: just copy the "title" column into the "new filename" column.

rmcanany, 2024-05-05:
OK.  Good deal.  I'm going to close this issue for now.  Feel free to reopen it if needed.

rmcanany, 2024-12-10:
Ready for the next version.  Thank you for the suggestion!  Closing as complete.

rmcanany, 2024-12-10:
Closing for real.

closed 2024-12-10 by rmcanany

farfilli, 2025-05-14:
Does the save part as part update the linked documents?

rmcanany, 2025-05-14:
You're talking about `Save model as` `Copy(*.*)`, right?  That doesn't check links.

farfilli, 2025-05-14:
Yes, I've been asked to implement the rename file (with update links) feature

rmcanany, 2025-05-14:
That should keep you busy!  Is that for your employer?  Should I add them as a Sponsor?

rmcanany, 2025-05-14:
Oh.  

I know you're not a huge fan of how the Task UI gets built.  I'm used to it now and kinda like it.  Anyway, if you tell me what controls you need, I can add them for you.

rmcanany, 2025-05-14:
To save you some typing, for `Where Used` you might be able to directly use `UtilsTopLevelAssembly.GetLinksTopDown`.  It uses Structured Storage, so it's fairly quick.  Here's the [Help Topic](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/HelpTopics.md#top-down)

I assume you would want to cache the results.  The new `HCErrorLogger` has a timestamp that stays the same throughout the run.  I could easily pass that down to the `TaskLogger` if you wanted access to it for your cache file name.