rmcanany/SolidEdgeHousekeeper #85: Basic documentation for adding a new task under new heirarchy

ChrisNeCle, 2024-05-14:
Can we get a brief writeup of how to add a new task under the new structure? I've attempted to add a new task. I created a new task file based on another task and added it to PreferenceUtilities.vb. It shows up in the Edit task list dialog. I cannot get it to show up in the Tasks tab. I have a couple of tasks I would like to add, but I'd like to get them in cleanly using the new and improved structure. I looked through the rest of the project and I don't see any other logic building the tasks list, though I'm not usually a VB.Net user so I may be overlooking something.

You can see what I've done in my RegenFlatModel branch if it helps: https://github.com/ChrisNeCle/SolidEdgeHousekeeper/tree/RegenFlatModel

rmcanany, 2024-05-14:
Adding it to BuildTaskListFromScratch() like you did is what I would have suggested.  As you know, I started the task list updates but got interrupted and haven't finished.  I'm dying to do so, but it will be a while, unfortunately.  

In the mean time, when I get a chance I'll dig around and see if I can at least get you going.

rmcanany, 2024-05-14:
What happens if you delete task_list.json in the Preferences directory?

ChrisNeCle, 2024-05-14:
That seemed to do the trick. It's working now. I'll put together another PR once I can get through some more testing. I can also start an issue for the task list not updating between updates if that would be helpful.

rmcanany, 2024-05-14:
Sounds good.  Looking forward to seeing your handywork!

Yes, please add an issue about the task list not updating.  It shouldn't be hard to fix, but might be hard (for me) to remember.

rmcanany, 2024-06-04:
No, that's where I left off.  It lets you make changes on the form, but does not yet update the real task list.

rmcanany, 2024-07-07:
Getting back to the original request, I added a quick note in Form1.vb, just below the Property Declarations.  It's more of a placeholder for now, but if you think of anything I should add, please don't hesitate to let me know.

rmcanany, 2024-07-11:
Closing as complete.

closed 2024-07-11 by rmcanany