rmcanany/SolidEdgeHousekeeper #86: Task list not updating between builds

ChrisNeCle, 2024-05-14:
Adding a new task and updating the task list in BuildTaskListFromScratch() will not update the task list if task_list.json exists in the target Preferences directory.

rmcanany, 2024-07-06:
I think this is now fixed with the `Edit Task List` functionality.  The source panel gets its Task List using `BuildTaskListFromScratch`, so everything should be available.  If you concur, I'll close this a complete.

farfilli, 2024-07-08:
@rmcanany for some reason the border drawn with UCTaskControl_Paint doesn't appear, do you know what could have broken it? In my opinion, this border is needed to distinguish better between tasks

rmcanany, 2024-07-08:
I appreciate the feedback.  Should be fixed now.

I'm pretty sure there's a gremlin in there somewhere.  Yesterday I had to change all the Form fonts from Microsoft Sans Serif back to Segoe.

farfilli, 2024-07-08:
@rmcanany If you have a look at the UCTaskControl_Paint event I have tweaked it to paint only the Top border.
With your modification, all the TLP borders are drawn which is not what I intended.

farfilli, 2024-07-08:
@rmcanany I fixed it with pull request #110

rmcanany, 2024-07-11:
Ok, thanks for that.  It's perfect!  Closing as complete.

closed 2024-07-11 by rmcanany