rmcanany/SolidEdgeHousekeeper #90: Property Find/Replace on FOA

rmcanany, 2024-05-16:
https://community.sw.siemens.com/s/question/0D5KZ000006oz1C0AQ/solid-edge-housekeeper-v20241-released

Issue with replace string: %{System.File Name (no extension)}.  Works on regular assemblies.

Probably need to fix Task_Common.SubstitutePropertyFormula()  SEDoc.FullName returns the name with the member name appended with the delimiter "!"

farfilli, 2024-05-17:
Robert, remember that each member of FOA has its own properties, there should be an option or a warning about that.

rmcanany, 2024-05-17:
Oh, that's news to me.  I appreciate the heads-up.  How are those accessed?  Activating each member in turn, or something like that?

farfilli, 2024-05-17:
Yes, activating each member gave you access to member unique properties; I don't know if in the latest releases a better method where implemented.

Programmatically, I think you need to reference the member by the "filename.asm!membername" sintax

rmcanany, 2024-05-17:
OK, thanks.  I'll check it out.

rmcanany, 2024-06-13:
I'll give that some thought.  Open to your suggestions, too.

rmcanany, 2024-07-16:
@farfilli 

I fixed the SEDoc.FullName issue and also added processing for FOA members.  I decided that it should only process those files if `Apply edits to all members` is enabled.  There is nothing to stop updates if it isn't; it was more to alert the user that each member will get the same treatment.  There's must be a way to treat each member individually, but I thought I'd wait until someone asked before putting in any time on it.

If you don't mind, I'll take you up on the offer to handle the structured storage piece.  Not sure the `Apply edits` is visible from there, but that method is marked as experimental, so whatever you decide should be fine.

One thing I'm not too fond of is the new `Project` PropertySet.  It's one more hurdle for the user and it seems like it wouldn't be that hard to determine where to look for those programmatically.  Thoughts?

farfilli, 2024-07-17:
About the editing of FOA files, a way could be to expand them in the file list and present them as if they were independent files.
I do not use FOA due to its limitations and behaviors, it's been years since I've asked for a concrete revisit of them; therefore if there aren't requests I would not implement them in Direct Edit.

About the Project thing, I would have preferred to keep them separated, anyway it's done now with #116

rmcanany, 2024-07-17:
OK, sounds good.  Thank you.  Closing as complete.

closed 2024-07-17 by rmcanany