rmcanany/SolidEdgeHousekeeper #91: Ability in property editor to use VB sintax for text manipulation

farfilli, 2024-05-22:
Example:

%{Title} = "PIPE L=1000 mm"
%{Title edited} = %{Title}.Replace("=",":")

The result would be:
%{Title edited} = "PIPE L:1000 mm"

Methods to be supported:
.SubString(a,b)
.Replace(a,b)
.IndexOf(a)
.LastIndexOf(a)

rmcanany, 2024-05-22:
That's a great idea.  It's on the list!

farfilli, 2024-06-05:
I have started experimenting with this implementation and realized that the command to be supported and the nesting level of each command could be a bigger project than I thought.
The best is to find an existing package that takes care of it; I have lurked a bit for a text parser and haven't seen (yet) one that satisfies me.

farfilli, 2024-06-05:
@rmcanany I think I found a great package to enable an advanced string manipulation; have a look at:
https://github.com/panoramicdata/PanoramicData.NCalcExtensions

My idea is to insert a button to create the "expression" and test it before launch the property editor

I will work on it in next days, you can assign this task to me

rmcanany, 2024-06-05:
I like it!

rmcanany, 2024-06-09:
That is awesome!  I'll do more testing when I get a chance.  I'll let you know if I run into anything.

I really like how you integrated this (and the StructuredStorage code) into Housekeeper.  Very clean and easy to follow.  Nice!

I did have one idea.  We use a simulation tool called JaamSim.  Its input editor (similar to yours) preserves carriage returns between edits, which helps readability.  NCalc probably wouldn't like that, but maybe we could do the cleanup just before the call to Evaluate().  What do you think?

farfilli, 2024-06-09:
Glad you like it.
I'm evaluating a different control for code editing, my idea is to integrate the following features:
- Wordwrap
- Better code highlighting 
- Auto complete of the code while writing 
- Code folding

rmcanany, 2024-06-09:
You're a MACHINE!

farfilli, 2024-06-10:
Implemented the following features:

- Wordwrap
- Better code highlighting
- Code auto complete
- Comments in code

rmcanany, 2024-06-10:
Good deal.  I'll check it out later this week.

rmcanany, 2024-06-12:
Closing as complete.

rmcanany, 2024-06-12:
Closing for real this time.

rmcanany, 2024-06-15:
That's a great addition.  I'll use it a lot.  I like the new examples, too -- very clear and helpful.

Speaking of examples, is there a reason to store `SavedExpressions.txt` in the Housekeeper root directory?  If not I'm going to change it to the Preferences directory.  Just FYI there's a method, `PreferencesUtilities.GetPreferencesDirectory()`, to get that location directly.

Another question -- do we still need the TextBoxes on the PropertyEditor to be multiline?  The display can be kinda confusing.  I pasted the expression "`toUpper('%{System.Title}')`", but only "`('%{System.Title}')`" was shown.  It took me a while to realize it was wrapping.  I figured you were trying to preserve the formatting, but in the couple of expressions I checked, I didn't see any `NewLine` characters.

Regarding the issue of updating material density, facestyle, etc., what would you think about disabling `Direct Edit` if that option is chosen?

farfilli, 2024-06-16:
Saved expressions.txt was just randomly placed cause I didn't investigate how the files are handled, you can move it and also insert more examples.

TextBoxes are multiline because I wanted the control to fit the TLP cell, probably there is a better method to draw them.

Material can disable the direct edit ofcourse

farfilli, 2024-06-24:
There is one thing that is not clear, we can add more examples.
Besides all the text manipulation tools, NCalc can solve mathematical expressions, which is its primary scope

rmcanany, 2024-07-06:
@farfilli  I moved `SavedExpression.txt` to the `Preferences` directory.  The file gets regenerated if it is not present.  

I tested the examples and they all seemed to work.  I was also able to add a new example.  I don't think I broke anything, but if you find something wrong, please let me know.  

As far as I know, this is now complete.  Let me know if you want to keep it open for any reason.

farfilli, 2024-07-08:
@rmcanany I don't see your example, are you sure it's been committed? Also, the txt source file could be moved to the project folder.

You can mark this as complete anyway

rmcanany, 2024-07-08:
@farfilli Yeah, I just did it for a test.  I didn't add it to the project.  

FYI, I moved the text for SavedExpressions.txt into the source code.  It's in PreferencesUtilities.CreateNCalcSavedExpressions() if you ever need it.  It's a pain to update it there, but makes for one less file I have to remember when creating a new release.  

Thank you for all your work on this.  It's a great addition!  Closing as complete.

closed 2024-07-08 by rmcanany