rmcanany/SolidEdgeHousekeeper #92: Direct File properties manipulation

farfilli, 2024-06-03:
Solid Edge files use the Microsoft COM Structured Storage technology as the basis for the Solid Edge file types.

Essentially Structured Storage allows an “internal file system” to be created within a single file, where the terms 
Storage and Stream correspond to folder/directory and file respectively.  Such files are often called Compound Files.

For a brief overview on Microsoft COM Structured Storage, visit the following Wikipedia site: 
https://en.wikipedia.org/wiki/COM_Structured_Storage 

I have been able to edit File Properties on a Solid Edge file and the main benefits are:
- No need to have Solid Edge installed
- No need to open\save\close a Solid Edge file
- I've been able to edit and successfully save the title properties in 41 ms

I can implement this function, it must be triggered in case only file properties manipulation actions are invoked; If the file must be opened for other tasks there is no need to use this alternative file properties method.

rmcanany, 2024-06-03:
That would be slick!  Go for it!

farfilli, 2024-06-03:
I will add the Sub that does the work, I let you add it to the main processing code, is that ok?

rmcanany, 2024-06-03:
Sure thing.  I've been wanting to do it anyway.  There are a few tasks, like `CheckLinks`, `MissingDrawing`, and a couple others that don't need to open the file in SE to run.  It'll be a while, though.  I'm swamped with work at the moment.  Hopefully I can get back to it later in the summer.

farfilli, 2024-06-03:
I'll have a look at those task and see if able to replicate the methods

rmcanany, 2024-06-03:
Sounds good.  The only one I can think of that accesses properties is checking the part number.  

Checking links only needs DesignManager.  MissingDrawing also needs DM if the drawing doesn't have the same name as the model, otherwise it can be done with OS tools.  I can handle those.

farfilli, 2024-06-04:
Completed with Pull request #94

farfilli, 2024-06-04:
I must say that updating file properties without Solid Edge seems SuperCharged, it is so fast! 🥇

ChrisClems, 2024-06-05:
If you know of any resources where it's documented, I'd be very interested in any information on the rest of the structured storage data formatting for SE files.

I poked around, and it looks like there's parasolid model data available. That would be huge for extracting models from SE documents without having to open them and run SaveAs() every time. You could spit out a folder of STEP models for a huge assembly in seconds.

[image]

In the example there, it appears there are two objects for the design model, two for the flat pattern model, and one for a sketch. Possibly a 2D of the flat pattern model? The extracted data looks like it's in the structure of an X_B binary parasolid file, but it won't open in Solid Edge. There is a P_B and a D_B stream for the non-sketches. P_B has GUISE=transmit_partition; in the header while D_B has GUISE=transmit_deltas; in the header.

When I have time, I think I'll develop a much simpler part to test on and see if any other CAD libraries will recognize them.

farfilli, 2024-06-05:
Did you have a look at the document posted some days ago by David Merret? It explains pretty well some aspect of the Solid Edge file format.
In that document another tool is mentioned that give a better view of the structured file

ChrisClems, 2024-06-05:
> Did you have a look at the document posted some days ago by David Merret? It explains pretty well some aspect of the Solid Edge file format. In that document another tool is mentioned that give a better view of the structured file

I just found it. This is amazing. I have weeks of reading and tinkering ahead of me now. This is likely going to be a game changer for the CAM software I'm working on.

closed 2024-06-15 by rmcanany

ChrisClems, 2024-06-20:
> > > Did you have a look at the document posted some days ago by David Merret? It explains pretty well some aspect of the Solid Edge file format. In that document another tool is mentioned that give a better view of the structured file
> > 
> > 
> > I just found it. This is amazing. I have weeks of reading and tinkering ahead of me now. This is likely going to be a game changer for the CAM software I'm working on.
> 
> Please share your discoveries. It would be great to gain access to the variable table and also, as you mentioned, being able to extract parasolid models.

I'm slowly churning my way through the data layout but I'm raising more questions than answers.

The parasolids are in binary partitions and binary deltas. I believe the partitions are "containers" that you'll find in a x_b file containing models and the deltas represent various states of the model that can be applied to roll to features. I can only find a single reference to the delta and partition data structures for parasolids but it's unfortunately a passing mention and wasn't terribly helpful. I spent a few hours looking for a way to open the partition file or convert it to another format. So far, no known tools are recognizing it. I can see the ASCII strings that represent the entities in the parasolid, but since it's binary that's about all I can read.

The project I'm currently working on is changing the document type specific options. These are the tabs in the options menu that show up based on what type of document you have open. I recently learned that these are not saved in your preferences, but instead are document-specific and saved in the file itself. There's no API exposed to change them. I have a use case to change the Simplify bSplines options on a large set of files and update the flat patterns. It would take weeks to do all of these by hand. The document-specific options for a sheet metal document are stored in the PSMcluster0 stream. It appears to be some kind of tabular data layout. I can find what appear to by the setting labels and units of measure but not the values. It's quite a lot of data to sift through and I'm a bit out of my element at this point. You can see the extracted streams attached to this issue: https://github.com/ChrisClems/EdgeSharp/issues/13

ChrisClems, 2024-06-20:
If you want to vote on the SE question here it would be helpful. Hopefully I can uncover some underlying data structure they're using: https://reverseengineering.stackexchange.com/questions/33041/trying-to-decode-a-data-structure-from-a-com-structured-storage-stream

rmcanany, 2024-06-20:
@ChrisClems  Wow!  You are persistent!