rmcanany/SolidEdgeHousekeeper #95: Integrate Direct File Properties Manipulation

rmcanany, 2024-06-04:
https://github.com/rmcanany/SolidEdgeHousekeeper/issues/92

farfilli, 2024-06-04:
@rmcanany Can you explain me the usage of this part of code? I need to implement an SEDoc free for the SolidEdgeRequired = false version

```
'######### TBD: Create an Overloads of TC.SubstitutePropertyFormula that uses the OLEProperties object
'If Proceed Then
'    Try
'        FindString = TC.SubstitutePropertyFormula(SEDoc, FindString, ValidFilenameRequired:=False)
'    Catch ex As Exception
'        Proceed = False
'        ExitStatus = 1
'        s = String.Format("Unable to process formula in Find text '{0}' for property '{1}'", FindString, PropertyName)
'        If Not ErrorMessageList.Contains(s) Then ErrorMessageList.Add(s)
'    End Try

'    Try
'        ReplaceString = TC.SubstitutePropertyFormula(SEDoc, ReplaceString, ValidFilenameRequired:=False)
'    Catch ex As Exception
'        Proceed = False
'        ExitStatus = 1
'        s = String.Format("Unable to process formula in Replace text '{0}' for property '{1}'", ReplaceString, PropertyName)
'        If Not ErrorMessageList.Contains(s) Then ErrorMessageList.Add(s)
'    End Try
'End If
```

rmcanany, 2024-06-04:
Yes.  That replaces a property formula with its value.  For example %{Custom.Engineer} -> "FRED".

rmcanany, 2024-06-04:
I looked at the code more closely.  A more accurate description is that it replaces all formulas in a string.  Here's a better example:
"Material: %{System.Material}, Engineer: %{Custom.Engineer}" --> "Material: STEEL, Engineer: FRED"

farfilli, 2024-06-05:
Completed with #96

farfilli, 2024-06-08:
System.Document Number
System.Revisiom Number
System.Projct Name

Those properties need to be implemented cause they are in a different stream.
Will do it soon

rmcanany, 2024-06-08:
OK.  Sounds good.  I'm going to try it out tomorrow.

While you're in there could you add some sort of error message for index-reference properties?  That's what I call them, anyway -- the ones with the `|R1`, `|R2`, ... modifiers.  I believe you said they can't currently be processed with the new Structured Storage technique.

rmcanany, 2024-06-09:
That is blazing fast!  Nice work!

farfilli, 2024-06-10:
Implemented Project properties (ReadOnly at the moment)
Implemented error handling for links with the syntax |R1 ecc

rmcanany, 2024-06-10:
OK, cool.  Should I close this issue, or leave it open for the upstream ReadOnly fix?

farfilli, 2024-06-10:
Let's wait some days and see if it get quickly fixed

farfilli, 2024-06-12:
When and if it get fixed I should be able to also update Material properties and potentially also FOA properties

farfilli, 2024-06-14:
With commit #105 I have completed the editing of Project Properties which works perfectly now.
I did investigate the ability to change other properties that are not text, for example, the material values and the hardware checkbox, there are still some problems so we are not done yet.

I have also started a revamp of the PropertyEdit page, choosing from System to Custom will make the PropertyName a ComboBox or a TextBox depending on which is selected.

The idea is to populate the combo with only usable properties
I have inserted now Project and System as populated, perhaps is better to just list all of them in a single category System? The code will then open the proper stream to edit

Will need to load proper values at the startup
[image]

farfilli, 2024-06-14:
I'm able to change properties that are number, used in material properties, Density is not there so it is not possible with this tool.
Does it make sense to enable material editing?

Is Hardware property manipulation of any interest? I can add it

farfilli, 2024-06-14:
Property edit page buttons work and also restoring properties on load.
If a row is left empty subsequent rows, even if compiled, seem to be ripped off, I don't know if it was like that before. Is this an acceptable behavior?

rmcanany, 2024-06-14:
That looks great!  Especially the ComboBox feature.  I might look at extending it to read a user's templates for custom properties.

Speaking of custom properties, I only added the System and Custom flags because I noticed that SE allows a user to name a custom property using a name that's already in another stream.  Not sure why they allow that.  I believe I read somewhere that causes problems with Callouts, etc.

I don't remember if it previously handled blank rows correctly.  My code for that is convoluted.  As I recall, it works by reading the form contents then populates a dictionary from that.  I realized in working on the Edit Task List functionality that a better approach is to first update the dictionary, then redraw the form with updated contents.  I need to do the same with Property Edit and Property Filter.

As far as material properties go, I added that because changing the property Material does not automatically update density, facestyle, etc.  I think I had to use Material Table functionality to do that.

Anyway, it looks good.  I might get a chance this weekend to play around with it.  Thank you for all your work!

farfilli, 2024-06-15:
The solution is to create a custom control that handle the property row, it will contains all the logic to get and set values. Main form will contain a list of them very easy to add , move and delete.

The code would became much more compact and clear.

And best of all you don't need all that dictionaries 😁

rmcanany, 2024-06-23:
@farfilli 
I'm testing the new `TaskControl` setup and I ran across a couple of issues with Edit Properties.

First, it doesn't seem to obey the `Edit outside Solid Edge` flag.  If that option is unchecked, it still uses the `StructuredStorage` method.  Also, if I select another task that cannot be done without SE, it generates an error message about a conflict.

Second, I used `Add folder and subfolders` to select files to process.  In the picture, I chose 7481-10000.  Some files in that tree have links to 7481-01000.  They shouldn't be processed, but Edit Properties tried anyway.  It couldn't find them because they're outside the folder I specified, so generated some errors.  

[image]

farfilli, 2024-06-24:
@rmcanany I just had a quick test and the "Edit outside Solid Edge" seems to work correctly; if selected it uses the Process(FileName as string) method if not it uses the Process(SeDoc, Configuration....) method.

The conflict error message appears when tasks of different types are selected; when a file is opened in Solid Edge there is no need for the structured storage method, also it could not work due to the file lock.

Perhaps a method that automatically unchecks the direct edit in case of conflict could be implemented.

About the folder question I'm not sure of what you mean, I have implemented a direct edit process method, it just work on the file list provided as usual

rmcanany, 2024-07-15:
@farfilli I can close this as complete, right?

rmcanany, 2024-07-18:
@farfilli 
I'm doing the documentation on this now.  I made some guesses on Direct Edit and Expressions.  If you get a chance, please look it over for errors/omissions. 
https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/README-2024.2.md#edit-properties

farfilli, 2024-07-19:
I had a quick look and there are some refuses:
- The project properties are read/write
- The expressions are implemented for replace text not for find text.

Do you think is needed for fing text as well?

rmcanany, 2024-07-19:
Updated the documentation.  I had started to remove references to the Find string, but then wasn't positive about it.  I don't think we need it.

farfilli, 2024-07-19:
If you think Expressions is better than RegEx we can implement it for find string too.
Should be an easy task and make it consistent.

rmcanany, 2024-07-19:
I do use RegEx occasionally, so would want to keep that.  If you have time to add EX for Find strings, I wouldn't object, but I wouldn't object to waiting either.  Your call.  Anyway, this is all working as originally envisioned so I'm closing for now.

closed 2024-07-19 by rmcanany

farfilli, 2024-09-02:
I would put emphasis on instruction page about the performance of Property Direct Edit

rmcanany, 2024-09-02:
Will do.