rmcanany/SolidEdgeHousekeeper #107: Request for new features

lrmoreno007, 2024-06-18:
When large assemblies are imported there are a series of jobs that are very repetitive, I have asked Siemens on several occasions to do it directly in SolidEdge as an option but they ignore it.

These would be the two new features?
1º.- Optimize the quality of the model (Part and Sheet Metal):
Inspect tab→Evaluate group→Optimize→→→ Full body →→→Finish

2º.- Hole recognition:
In the part environment, the command resides on the Home tab→Solids group→Hole command list.
In the sheet metal environment, the command resides on the Home tab→Sheet metal group→Hole command list.
In the assembly environment, the command resides on the Features tab→Assembly Features group→Hole command list. **** In assembly mode, it is never really needed when importing a model.

It's always do the same work, over and over again always the same. Please, would it be possible to introduce these new features? I'm trying to program it myself and that's how I came across this project, but I don't really have the programming knowledge to do it.

Thanks in advance.

rmcanany, 2024-06-18:
Thank you for the suggestion.  It's on the list!  

No promises on implementation.  It's not a function I really need.  If you do get something working, assuming you need batch processing, I'd be glad to help you get it functioning as a [Housekeeper External Program](https://github.com/rmcanany/HousekeeperExternalPrograms#readme).

rmcanany, 2024-07-15:
Just wondering if you made any headway on this.  There was a related post on the Solid Edge forum a couple of weeks ago.  Here's the link:
https://community.sw.siemens.com/s/question/0D54O000085AlRxSAK/recognize-holes-function

lrmoreno007, 2024-07-15:
Yes! I'm User16370554015158121216 in SolidEdge Community. 
I managed to make a macro with what I wanted, this is already progress for me but I still had to go file by file optimizing.  So I decided to take a look at your source code and I did it successfully too.  Now I just need some time to share it with you.  Maybe later you could debug it a little and put some options as a selectable option.
Thanks.

rmcanany, 2024-07-15:
Oh, that's great news!  You won't be sorry taking the time to learn how to automate SE.  

When it's to a point you want me to look it over, just zip it up and attach it here.  I'd be glad to help out!

lrmoreno007, 2024-07-15:
Ok! Tomorrow I will send you

lrmoreno007, 2024-07-16:
[SolidEdgeHousekeeper-lrmoreno007.zip](https://github.com/user-attachments/files/16250568/SolidEdgeHousekeeper-lrmoreno007.zip)

Check it out

rmcanany, 2024-07-16:
Got it!  I'll have a look, hopefully later this week.  

I saw on your GitHub page you're into Arduino.  Except for dealing with the horrible API documentation, you'll have no trouble automating Solid Edge.

lrmoreno007, 2024-07-17:
I have created a  Pull Request #115.

> Got it! I'll have a look, hopefully later this week.
> 
> I saw on your GitHub page you're into Arduino. Except for dealing with the horrible API documentation, you'll have no trouble automating Solid Edge.

The problem is that I am only a programming fan, I have done some more or less serious things in VB6 as my final project at the university (I'm Mechanic Engineer, the project was a program for matrix calculation structures) and correct some problems in a ad-hoc ERP in my work, all surely very poorly done hahaha. Also, as you say, I have done little things in Arduino, Python, Processing, Industrial PLCs, etc.
I thought VB.Net was going to cost me more but I got the hang of it. And there was the problem of the IDE too: I don't have Visual Studio but I found JetBrains Riders EAP, which is free.
Regarding the SolidEdge API, it is very poorly documented but with the help of the SDK, Spy SolidEdge and reviewing as you have done, I got it.

Take a look at the code and I'm sure you'll give it a more elegant and better done touch hahaha.

Thanks!!!

P.D.: By the way, forgive my English, I'm Spanish.

rmcanany, 2024-07-17:
Your English is great, so is your programming!  Thank you for the contribution.  Closing as complete.

closed 2024-07-17 by rmcanany

lrmoreno007, 2024-07-17:
I have created a new repository with the macro that I told you and I have set as an option that your project could be used for several files and large projects.

In next days i will make a new pull request to add the capacity of move the body from Ordered to Syncronous before carrying out the optimization process because otherwise the holes will not be recognized. This is already implemented in my macro.

On the other hand, the task has nothing in the help, let's see if you can do it based on what I have put in the Readme of my macro.

Thanks again.

rmcanany, 2024-07-17:
Good deal.  I like the idea to take care of the transition to Sync automatically.  Why bother the user with a chore you can do without too much effort?

I'm working on documentation right now.  I'll see what I can pull from the macro documentation into the task writeup.

rmcanany, 2024-07-21:
@lrmoreno007 
Documentation has been updated.  If you get a chance, please look it over and let me know of any errors/omissions.  

As I mentioned before, I thought about renaming the task.  It's now called Recognize Holes.  I also did a bit of code refactoring -- mostly to match some style conventions in the other tasks.  I hope you don't mind.

Based on some limited testing, I changed a couple of things...
-- Added a model count check.  I've run into issues when a file has no models present.
-- Changed `If CType(SolidEdgePart.ModelingModeConstants.seModelingModeOrdered, Boolean) Then` to `If tmpSEDoc.ModelingMode = SolidEdgePart.ModelingModeConstants.seModelingModeOrdered Then`
-- Enclosed `RecognizeAndCreateHoleGroups()` in a Try/Catch block.  That command was generating an exception on non-imported parts.  (I realize that type of file has no business in this command, but you never know.)
-- Removed the Hide Constructions/Fit View code.  That's available elsewhere in Housekeeper already.