rmcanany/SolidEdgeHousekeeper #109: Localized system properties

farfilli, 2024-06-24:
There is an issue with the System.Title and other "named" properties.
If you have a localized Solid Edge, those properties will be translated into the user language.
The actual method looks for the exact name and it will return a property not found error.

A possible solution would be to create a map SystemPropertyName --> PropertyIndex and use the index to get the correct property. This issue is present only when editing properties through Solid Edge; the structured storage method is not affected.

rmcanany, 2024-06-24:
Sounds like a good idea to me.  That must be a recent change.  I had a user early on who discovered he had to use English for those properties, even though they were presented in Polish in SE.

rmcanany, 2024-06-24:
I recently saw a list of system properties that SE actually supports.  Was that from you?  

Also, the property indices you mention, would those be the item number in whatever stream holds the property, or something else?

farfilli, 2024-06-24:
I have created the list in combo box for system properties, but not created a mapped index to use in the PropertySet
Structured storage is always in English so the problem occurs only when edit on Solid Edge

farfilli, 2024-07-19:
fixed with #118 
I haven't created a specific function that returns the property but wrote it directly inside the GetProp sub.
It may be useful for other usage to create a proper function.

rmcanany, 2024-07-19:
Did a quick test and all seems to be working.  I'll do some more before I close this out.

I added `Material` and `Sheet Metal Gage` to the checks in GetProp.  

In TaskEditProperties, at one point I have:
`If (PropertyName.ToLower = "material") And (PropertySetName.ToLower = "system")`
Correct me if I'm wrong, but since the ComboBox entries are in English, this should still work even if a user is running in Localized mode, right?

farfilli, 2024-07-19:
Yes, It works and you don't need the toLower statement at the combo box entries are not editable.

The problem is on the property name that SolidEdge APIs returns localized and therefore you need the mapping.

Material it's a strange beast anyway. Even if through API is present in the PropertySet it's not in the UI and not in the Structured Storage. Said that I don't know if it come localized or not, I have to check it but can't till next Tuesday.

farfilli, 2024-07-20:
It's gonna be a huge release the next one 😀🔝

rmcanany, 2024-07-23:
@farfilli When you get to it, let me know what you find about the language for Material, etc.  If they're localized, I have a few things to fix.

farfilli, 2024-07-23:
@rmcanany the property Material is not localized and it is in MechanicalModelling.Item(1)

rmcanany, 2024-07-23:
OK.  Thank you.  

Same with the rest of them, too?  (Density, Coefficient of Thermal Expansion, etc.)

farfilli, 2024-07-23:
That's a mix of translated and English.
I will send you a screenshot tomorrow, not in front of PC now.

I was thinking that for standard properties we could map them all, there will be no need to find them in this way.

rmcanany, 2024-07-23:
Yeah, mapping them sounds like better way to go.  I'll look into it.

farfilli, 2024-07-24:
I have highlighted the localized properties; by the way density is not part of the properties but its an exposed variable
[image]
[image]

rmcanany, 2024-07-24:
Very helpful.  Thank you!

rmcanany, 2024-07-29:
@farfilli @ChrisClems @lrmoreno007 

Hey All, I'm going to take a look at this today.  Once this one is fixed, I think we're ready to do a release.  I have been working on the [release notes](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/release_notes.md).  When you get a chance, please look over your sections and let me know if I got anything wrong or left anything out.  Thank you!

farfilli, 2024-07-29:
@rmcanany the link on DirectEdit leads to the wrong position. 

Thoughts:
To be able to implement the expression editor I have used those 2 components:
- [CoreCLR-NCalc](https://github.com/sklose/NCalc2)
- [FCTB](https://www.codeproject.com/Articles/161871/Fast-Colored-TextBox-for-syntax-highlighting-2)(FastColoredTextBox)

To be able to implement the Direct edit of properties I have used this component:
- [OpenMcdf](https://github.com/ironfede/openmcdf)

I would check on license terms if you need to mention them on the ReadMe; I would mention them anyway they are all great components

rmcanany, 2024-07-29:
Good points, especially about the external components.  I'll get that fixed.

rmcanany, 2024-08-04:
Superseded with https://github.com/rmcanany/SolidEdgeHousekeeper/issues/123.  Closing.

closed 2024-08-04 by rmcanany