rmcanany/SolidEdgeHousekeeper #112: Auto generate draft of flat

ChrisClems, 2024-07-11:
Today I had a burning need to generate a bunch of drawings of flat patterns in a PDF to review some geometry placement on 600+ parts at a time. I got to a very rough but very working implementation today and thought it might be a good candidate to include as a new feature. I know my engineers are going to be over the moon about it. If we think it's a good fit I can start work on it whenever my current project load settles down. Once I get all the C# written for all the features I want, I can translate it to VB if I can get some help with the UI side of things.

I still need to run it through a few more assemblies without the shortcuts I took to ship the product today to weed out any issues. The bones are working though.

C#
```
using System.Net;
using System.Runtime.InteropServices;
using EdgeSharp;
using EdgeSharp.Extensions;
using SolidEdgeAssembly;
using SolidEdgeDraft;
using SolidEdgeFramework;
using SolidEdgePart;

internal static class Program
{
    // TODO: Get rid of this DllImport garbage and find a more elegant way to get the progID
    [DllImport("ole32.dll", CharSet = CharSet.Unicode, PreserveSig = false)]
    static extern string ProgIDFromCLSID([In()]ref Guid clsid);
    
    [STAThread]
    private static void Main()
    {

        var app = AppHelper.Connect();
        
        const ApplicationExtensions.BackgroundOptions backgroundOptions =
            //ApplicationExtensions.BackgroundOptions.Invisible |
            ApplicationExtensions.BackgroundOptions.HideAlerts |
            //ApplicationExtensions.BackgroundOptions.NonInteractive |
            //ApplicationExtensions.BackgroundOptions.NoScreenUpdating |
            //ApplicationExtensions.BackgroundOptions.DelayCompute |
            ApplicationExtensions.BackgroundOptions.None;

       app.SetBackgroundOptions(backgroundOptions);

        var asmDoc = (AssemblyDocument) app.Documents.Open(@"E:\path\to\assembly.asm");
        
        asmDoc.TraverseAssemblyWithAction((doc) =>
            {
                if (doc.Type == DocumentTypeConstants.igSheetMetalDocument)
                {
                    var sheetMetalDoc = (SheetMetalDocument)doc;
                    // Skip docs with no flat pattern
                    if (File.Exists($"{sheetMetalDoc.Path}/{sheetMetalDoc.Name}.dft")) return;
                    var flatPatternModels = sheetMetalDoc.FlatPatternModels;
                    if (flatPatternModels.Count < 1) return;
                    var flatPatternModel = flatPatternModels.Item(1);
                    var flatPattern = flatPatternModel.FlatPatterns.Item(1);

                    double minX;
                    double minY;
                    double maxX;
                    double maxY;
                    try
                    {
                        // Get range of flat pattern.
                        flatPattern.Range(out minX, out minY, out _, out maxX, out maxY, out _);
                    }
                    catch (NullReferenceException)
                    {
                        // Don't know why this happened.
                        // TODO: Find out what is throwing null exceptions on FlatPattern.Range()
                        return;
                    }
                    
                    // Calculate total height and width of flat pattern X/Y
                    var xRange = maxX - minX;
                    var yRange = maxY - minY;

                    // Convert to inches
                    xRange = xRange * 39.37;
                    yRange = yRange * 39.37;

                    // Calculate X and Y scale required based on max size you would like the drawing view
                    var xScale = 16 / xRange;
                    var yScale = 7 / yRange;
                    // Scale needed is whichever is lower to prevent overflow from desired max window size.
                    // Shrunk by a tiny bit to accommodate drawingviews with weird centroids.
                    var scale = Math.Min(xScale, yScale) * 0.95;
                    
                    // This should be unnecessary. Bake it into EdgeSharp.
                    Guid guid = typeof(DraftDocument).GUID;
                    string progId = ProgIDFromCLSID(ref guid);

                    // Create draft doc and add SheetMetalFlatView in top orientation.                    
                    var draftDoc = (DraftDocument)app.Documents.Add(progId);
                    var modelLinks = draftDoc.ModelLinks;
                    var modelLink = modelLinks.Add(doc.FullName);
                    var sheets = draftDoc.Sheets;
                    var drawingSheet = sheets.Item(1);
                    var drawingViews = drawingSheet.DrawingViews;
                    // x and y will be where the centroid of the DrawingView is placed relative to x/y 0 of the sheet.
                    // 0.2159 x 0.15 will place the center of the DrawingView at the center point of a default
                    // 11" x 17" size sheet, offset slightly in +Y to compensate for a standard background
                    // bottom header. Find a better way to handle this by finding sheet size API.
                    var sheetMetalDrwaing = drawingViews.AddSheetMetalView(modelLink,
                        ViewOrientationConstants.igTopView, scale, 0.2159, 0.15,
                        SheetMetalDrawingViewTypeConstants.seSheetMetalFlatView);
                    draftDoc.SaveAs($"{sheetMetalDoc.Path}/{sheetMetalDoc.Name}.dft");
                    app.DoIdle();
                    draftDoc.Close();
                    // Stop
                }
            }, true
            );

    }
}
```

This code does the following:

1. Walks and assembly recursively, looking for sheet metal parts with flat patterns.
2. Calculates the height and width of the flat pattern.
3. Gets the size of the area which you'd like the drawing view to be.
4. Scales the drawing view to fit within the area.
5. Places the drawing view at a chosen point on the sheet.

Planned additions:
* Better handling of drawing view placement on sheet / handling different size sheets.
* Rotate drawing views so longest edge is always on X for landscape sheets.
* Handle custom templates, backgrounds, etc. (Can a user make a custom template that notes the default center point of the drawingview to be consumed by the script or better yet, a window?)
* Metadata to drawing (how much of this does SE automate? I've barely touched the draft environment before today)
* PDF output
* Big boy error handling.

rmcanany, 2024-07-12:
That sounds like a useful feature.  I can handle the UI part.  Not sure we need the assembly-traversal or PDF output, as that's already part of Housekeeper and might be confusing.  

If it's any easier, once you get it working like you want, you could simply supply the source code and I can do the conversion/integration on my end.

ChrisClems, 2024-07-15:
> That sounds like a useful feature. I can handle the UI part. Not sure we need the assembly-traversal or PDF output, as that's already part of Housekeeper and might be confusing.
>
> If it's any easier, [...]

Oh yeah I wasn't suggesting the assembly traversal. That's just the prototype code I'm working with for my implementation. I'm going to try to get rotation baked in, as well as an option to do two drawings (a back side and a top side) on the same sheet while scaling it then I'll throw together something more practical for Housekeeper.

ChrisClems, 2024-07-22:
I'm going to drop the current C# code here. I won't have time to translate it any time soon but if you end up with some free time, this should get things started. It's getting the flat pattern view drawing, adding it to a sheet, detecting if it's taller in Y than in X and if so rotating 90 degrees, then using the DrawingView range to find the center and place it based on the sheet center.

It should probably better support non-default sheet sizes. It should support .par documents (not implemented in my code yet as all my flat parts are sheet metal). I really need to break it down into more manageable functions before I start expanding the features. PDF output should be optional. Now that scaling and centering is solid using the DrawingView range, it should be fairly trivial to support other view types and orientations. Need to look at the constants available for drawing views and experiment to see what works well.

```c#
var sheetMetalDoc = (SheetMetalDocument)doc;
                    Debug.WriteLine($"Generating drawing for {sheetMetalDoc.Name}");
                    // TODO: Detect UpToDateness of drawing before moving on
                    if (File.Exists($"{sheetMetalDoc.Path}/{sheetMetalDoc.Name}.dft"))
                    {
                        Debug.WriteLine("Drawing exists. Skipping.");
                        return;
                    }

                    var flatPatternModels = sheetMetalDoc.FlatPatternModels;
                    if (flatPatternModels.Count < 1)
                    {
                        Debug.WriteLine("No flat models. Skipping.");
                        return;
                    }

                    var flatPatternModel = flatPatternModels.Item(1);
                    var flatPattern = flatPatternModel.FlatPatterns.Item(1);

                    double minX;
                    double minY;
                    double maxX;
                    double maxY;
                    try
                    {
                        // Doesn't take into account tabs. Only measures flat pattern object range.
                        //TODO: Try to get a range which includes tabs on the flat model
                        flatPattern.Range(out minX, out minY, out _, out maxX, out maxY, out _);
                    }
                    catch (NullReferenceException ex)
                    {
                        Debug.WriteLine(ex);
                        using StreamWriter logFile = File.CreateText(@"E:\2687\RTR-Labeled\err.log");
                        logFile.WriteLine($"NullReference while processing {sheetMetalDoc.Name}");
                        logFile.WriteLine(ex);
                        return;
                    }

                    var xRange = maxX - minX;
                    var yRange = maxY - minY;

                    // Check if drawing is taller than it is wide for rotation and scale handling.
                    var tallDrawing = (xRange < yRange);
                    
                    // 16" x 7" will get you good drawing size on a default 11"x17" sheet with a standard style background
                    var xScale = 16 / xRange.MetersToInches();
                    var yScale = 7 / yRange.MetersToInches();

                    // Rotate the scale if flat pattern is taller than it is wide.
                    // DrawingView will be rotated after placing.
                    if (tallDrawing)
                    {
                        xScale = 7 / xRange.MetersToInches();
                        yScale = 16 / yRange.MetersToInches();
                    }

                    var scale = Math.Min(xScale, yScale) * 0.95;

                    // This is stupid and unnecessary. Bake it into EdgeSharp.
                    Guid guid = typeof(DraftDocument).GUID;
                    string progId = ProgIDFromCLSID(ref guid);

                    // Get draft doc, model link, and sheet
                    var draftDoc = (DraftDocument)app.Documents.Add(progId);
                    var modelLinks = draftDoc.ModelLinks;
[... 27 lines cut ...]

                    // Some flat pattern drawing origins end up in inconsistent places
                    // Get drawing view center to properly place it centered on sheet using offset from sheet center
                    double drawingViewXMin;
                    double drawingViewYMin;
                    double drawingViewXMax;
                    double drawingViewYMax;
                    sheetMetalDrawing.Range(out drawingViewXMin, out drawingViewYMin, out drawingViewXMax,
                        out drawingViewYMax);

                    var drawingViewXMid = (drawingViewXMin + drawingViewXMax) / 2;
                    var drawingViewYMid = (drawingViewYMin + drawingViewYMax) / 2;
                    Debug.WriteLine($"Drawing center: {drawingViewXMid.MetersToInches()}, {drawingViewYMid.MetersToInches()}");

                    drawingViewOriginX += sheetXMid - drawingViewXMid;
                    drawingViewOriginY += sheetYMid - drawingViewYMid;
                    sheetMetalDrawing.SetOrigin(drawingViewOriginX, drawingViewOriginY * 1.1);

                    draftDoc.SaveAs($"{sheetMetalDoc.Path}/{sheetMetalDoc.Name}.dft");
                    app.DoIdle();
                    // Optionally save as PDF?
                    draftDoc.SaveAs($"{sheetMetalDoc.Path}/{sheetMetalDoc.Name}.dft.pdf");
                    app.DoIdle();
                    draftDoc.Close();
```

rmcanany, 2024-07-22:
Yeah, that's great!  I'll take it from here.  I'll leave this issue open for now, in case I run across something I don't understand.  Thank you!

rmcanany, 2024-07-27:
@ChrisClems 
I got something working today.  If you get a chance, please check it out and let me know anything I messed up or left out.

ChrisClems, 2024-07-29:
@rmcanany Looks great! Just threw a few dozen parts through it and it hasn't choked up. I added an X/Y offset with #121 but I think it needs some thought and work for unit conversions. I couldn't think of a good way to handle it off the top of my head without some major discussion about how you'd like unit conversions handled.

rmcanany, 2024-07-30:
I added offset unit checkboxes for "in" and "mm".  When you get a chance, check it out and let me know what you think.

rmcanany, 2024-08-04:
Closing as complete.  @ChrisClems Thank you!

closed 2024-08-04 by rmcanany