rmcanany/SolidEdgeHousekeeper #122: Update document type to match extension

ChrisClems, 2024-07-30:
I wasn't going to suggest this originally because of how horribly dangerous it sounds and because of external dependencies. Now that OpenMcdf is a dependency, that's one out of two of my reservations taken care of so I'll throw it out there.

There's a lot of misinformation floating around community forums about the PAR and PSM formats and how they're interchangeable. Many people seem to have followed that advice and it may not become apparent until months or years down the road when you try to feed that part to some CAM or other software that breaks with a cast or null exception because it's reading the type from the file extension instead of the document type.

I had a few dozen parts like this in my library. I forcefully changed the CLSID in the COM structured storage to match the file extension.  The parts open and scratching the surface of model manipulation, appear to be fully functional parts with no corruption. They report the correct new document type in the properties as well as allowing Save and Save As using the new document type. I suspect at one point there were differences but now it's all unified in the parasolid format.

The code is super simple:

```c#
    public static void ForceSaveAsSheetMetalDocument(this PartDocument doc, string newName)
    {
        // TODO: Error handling.
        var sheetMetalClsid = new System.Guid("dd8522e0-2375-11d0-ac05-080036fd1802");
        var originalDocPath = Path.GetFullPath(doc.FullName);
        var compoundFile = new CompoundFile(originalDocPath);
        var rootStorage = compoundFile.RootStorage;
        rootStorage.CLSID = sheetMetalClsid;
        compoundFile.SaveAs(newName);
    }
```

The reverse operation is the same with GUID `23c52e80-4698-11ce-b307-0800363a1e02`

Pros of including it:
* It's not an easily accessible thing for non-programmers to do so Housekeeper would be basically the only option
* We're already using the required dependency for the new properties manipulation.

Cons of including it:
* Very niche use case that few people will likely ever need.
* Unforeseeable potential for file corruption by force-changings things that aren't meant to be.
* If possible, it's better to remodel your parts as the correct file format from scratch.

Given the small code footprint I think it's a decent candidate for a task, albeit perhaps hidden behind warnings and not supporting overwriting files. Now that the task list is customizable, could it be hidden as a non-default option that one needs to deliberately enable before using?

rmcanany, 2024-07-30:
I wouldn't be opposed.  The only thing is, like you say, it's a pretty niche thing.  Would you consider making it an external program?  There's a separate repo, [HousekeeperExternalPrograms](https://github.com/rmcanany/HousekeeperExternalPrograms#readme), set up for such things.

I want to avoid having too many narrowly-focused commands in the main interface.  There are a couple in there already.  I plan to convert those to external programs eventually.

ChrisClems, 2024-07-31:
I like that idea. Having to seek out and connect an external program would also distance it from people who would run it while ignoring the warnings. Once I get some free time I'll dig through the examples and put something together to test.

closed 2024-07-31 by ChrisClems

rmcanany, 2024-08-02:
I realize you have closed this issue for now, but wanted to mention something anyway.

Working on localized properties yesterday, I ran across something relevant.  In sheet metal files, the Mechanical Modelling collection has a property called Sheet Metal Gage.  Part files don't.  

Structured storage uses English names for properties.  I have to map localized names because of that.  I'm currently using the item index of the property to do the mapping.  In sheet metal files, Sheet Metal Gage is item 2.  In part files, item 2 is Face Style.  If SE doesn't deal with that based on Clsid, it will screw up my mapping.