rmcanany/SolidEdgeHousekeeper #123: Update Property & Variable Edit and Property Filter

rmcanany, 2024-08-04:
-- Incorporate User Controls (Supersedes https://github.com/rmcanany/SolidEdgeHousekeeper/issues/106)
-- Deal with localized properties (Supersedes https://github.com/rmcanany/SolidEdgeHousekeeper/issues/109)
-- Incorporate new `TemplatePropertyList` functionality
-- Add Save/Retrieve/Delete for user inputs

farfilli, 2024-08-05:
@rmcanany "Add Save/Retrieve/Delete for user inputs" where is this located? I don't get it

rmcanany, 2024-08-05:
Oh, those are just planning statements.  I'm going to work on it this week.  I haven't uploaded anything yet.

farfilli, 2024-08-05:
Sorry, misunderstood

rmcanany, 2024-08-05:
Actually I think I was the one who misunderstood.  You asked specifically about Save/Retrieve/Delete.  That is like what you did with saved expressions, only for Find/Replace.

farfilli, 2024-08-05:
> Actually I think I was the one who misunderstood. You asked specifically about Save/Retrieve/Delete. That is like what you did with saved expressions, only for Find/Replace.

Yes that's it, I thought you had implemented yet; It's not mandatory but would be a nice addition beside the "Plans" for task lists

rmcanany, 2024-08-12:
@farfilli 
I added the lookup functionality using property item numbers as you suggested.  The mapping is *from* localized *to* English.  I thought that would be more intuitive for the users.  

I'm hoping you could run a test for me.

- Update properties from template (described [here](https://github.com/rmcanany/SolidEdgeHousekeeper#1-property-filter), scroll down a bit to see it)
- Customize the list (described in the same location)
- Run localized `Edit Properties` with the `Edit outside Solid Edge` option.  (Populate the list using the combo boxes.)

farfilli, 2024-08-13:
Hi Robert, this is an holiday week in Italy. I'll take a look by the middle of next week.

rmcanany, 2024-08-13:
Hey Francesco, that's cool.  Enjoy your time off!

farfilli, 2024-08-21:
@rmcanany I had a quick look and it seems the System properties list doesn't load on the combo box

[image]

rmcanany, 2024-08-21:
Hmm.  I'll check it out.  That's in localized mode, right?

Not sure I mentioned it, but that list is populated on the Templates Tab.  Any chance you didn't do that step?

[image]

farfilli, 2024-08-21:
I didn't check anything, and templates are not set at all. I would add a warning message or something to avoid this situation. I will retry again tomorrow

rmcanany, 2024-08-21:
Good idea.  Will do.

rmcanany, 2024-08-21:
That warning should be there now.  I added it for `Property Filter` and `Edit Properties`.

If you don't see it, try deleting the following files in the Preferences directory:
`form_main_settings.json`
`template_property_dict.json`
`template_property_list.json`

farfilli, 2024-08-22:
So I have added the templates and chosen the properties with the "UPDATE" button, which appears in both system and custom. This may lead to error and confusion by the user due to the fact that System properties will never have the custom ones

[image]

farfilli, 2024-08-22:
I strongly recommend to have two separate properties lists, one for the Systems and one for the Customs properties

rmcanany, 2024-08-22:
Actually, you don't need that.  It's not obvious, but the listed properties know to which category they belong.  Selecting the property sets the category.  That's in the help topic, but of course no one reads that.  

I need to come up with a way to make it obvious.  I could maybe disable the System/Custom drop down when the list is populated.  Or maybe I should just get rid of the category thing altogether.  It's only there because SE allows duplicate names, which is a bad idea in my opinion.  What do you think?

Oh, and back to the original question.  Did the localized mapping work with Structured Storage?

rmcanany, 2024-08-22:
I'll look into getting rid of the category (aka PropertySet).

If I understand your question, there isn't a separate category for direct edit.  You just select the `Edit outside SE` option.

farfilli, 2024-08-23:
Selecting a property doesn't correctly select the PropertySet
[image]

rmcanany, 2024-08-23:
That should be "Custom" I take it.  Could you provide a sample file for me to play with?  Maybe by email, or a message on the SE Forum.

farfilli, 2024-08-23:
I think the problem is with any custom property

rmcanany, 2024-08-23:
I tried it just now.  It's working for me.  Can you place a breakpoint and inspect TemplatePropertyDict to see what might be going wrong?

By the way, It doesn't currently update until the PropertyName gets a LostFocus event.  But that's the same for either System or Custom.

rmcanany, 2024-08-23:
No sweat.  Talk to you next week.

rmcanany, 2024-08-25:
I removed the use of System. and Custom. from the code.  I haven't updated the UI -- wanting to make sure it's likely to work before I start undoing a bunch of stuff.  

If you get a chance to test it I would welcome any feedback.  There are probably some bugs -- it has not been thoroughly tested by any means.

I haven't tried FOA or FOP with structured storage yet.  I thought I might look at that next.

farfilli, 2024-08-26:
I had a quick test and it seems to work both in interactive and in direct edit mode.
One thing to consider is how the %{System.Property} should be written in formula syntax; in direct edit, it is mandatory that is not localized; I guess those who want to use that syntax should do some tests before proceeding. Proper documentation should be written.

rmcanany, 2024-08-26:
OK, cool.  Thank you.

I build a lookup table for translating localized to English.  It gets populated when you run Update on the Templates Page.  It uses your idea to map by Item Number.  I added a check that the lookup table is populated for `Run outside SE`.

As I mentioned, `System.` and `Custom.` are going away.  They're no longer used in the code.  Once I do some more testing, I'll remove them from the UI.

That's a good point about the documentation.  I'll make sure to mention it.

farfilli, 2024-08-27:
I have this situation:
- A Custom property named "REVISIONE"
- I need to move its value to the System properly "Revision Number"
- System properties also contain a property named "Revisione"

How do I specify which property to change/read if system. and custom. are removed?

This kind of maintenance on file is where HK does its best so I think we should handle duplicated property names. In fact I need to move the Custom.Revisione to System.Revision Number to fix the annoying duplicate property message on all the archive

rmcanany, 2024-08-27:
Makes sense.  I'll add that back in.  I'm thinking I could make it optional.  Can you think of a reason not to?

farfilli, 2024-08-27:
I think it's a waste of code; Solid Edge users are familiar with the concept of custom properties and system(or we can call them standard?) properties therefore my first choice would be to leave them everywhere.

The autoselect of custom Vs system would be a nice thing but I would prefer 2 different combolist already filter accordingly to the selected group.

rmcanany, 2024-08-27:
Good points.  That's what I'll do.  Thank you!

farfilli, 2024-08-28:
One more thought: it could be that the system and custom are removed and auto determined accordingly and at the same time can also specify it manually

Example

Revision will automatically be the system one
Custom.Revision will be the custom one

It this way we can handle the duplicated properties when present

Sorry to not have provided it before

rmcanany, 2024-08-28:
That's more or less how I planned to do it.  If the user picks the Category first, I'll filter the list.  If they pick the Property first, I'll populate the Category.  

Duplicates will appear in both filtered lists.  I won't populate the Category for duplicates; I want to have the user make an intentional decision.

farfilli, 2024-08-28:
And how you determine the duplicates? By template?

rmcanany, 2024-08-28:
Yes, that was the idea.  It's also possible to add a property to the customized list.  I was thinking of adding a prompt for Property Set in that case.  

I could be missing something.  What do you think?

rmcanany, 2024-09-02:
Uploaded some fixes just now.  It isn't done.  I still need to filter based on category selection.

rmcanany, 2024-09-03:
@farfilli 
Filtering properties based on category selection should be working now.  Let me know if you run into any issues, especially with duplicate property names and/or running localized.

farfilli, 2024-09-04:
Property filter and category selections seems to work; however I experienced many missing system properties
[image]

rmcanany, 2024-09-04:
OK, thank you.  I think I know what's going wrong.  I'm filtering out a bunch of properties, like Number of Characters.  I bet I'm not using the lookup table in that section of the code.  I'll take a look today.

farfilli, 2024-09-05:
I done some test both direct and SE property manipulation and it seems to work fine.
I found only 1 problem, when deleting a custom property, if the Find Text and Replace Text is not empty due to a previous usage you got an error

[image]

When "X" is selected those fields should be ignored, or another approach would be to black them when they became read-only

rmcanany, 2024-09-05:
Should be fixed now.  Hopefully I didn't break something else in the process.  (I haven't added the `Customize Favorites` shortcut yet.)

I really appreciate your help on testing, by the way.

rmcanany, 2024-09-08:
@farfilli 

I'm having trouble adding a new property with Edit Properties outside SE.  

I added some breakpoints and it looks like the property is added.  However, the it doesn't show up in SE.  I can't see it with SSView either.  I added a Try/Catch.  The command `userProperties.PropertyNames.Keys.Max() + 1` was causing an exception.  I tried a couple of other ways, but I'm just guessing.

farfilli, 2024-09-09:
I have restored the .Max() + 1 method as it is the simplest way to get a unique valid ID.
Can you provide more details on the exception error you get?
I have tried with several files, the only problem I have encountered is on Family Of Assembly files; the properties storage has a different layout and it need a specific method to handle it.

farfilli, 2024-09-09:
I have updated the OpenMcdf DLLs with  #131 it is not a release but an improvement on the previous one.
It should provide more details in case of exceptions in properties editing.

farfilli, 2024-09-10:
Fixed with #132

rmcanany, 2024-09-10:
I don't get the error message anymore, but some files are still not getting the new property.  It appears to be those which have no custom properties at the outset.

I ran it on a copy of the SE2024 Training directory.  After the run, I set up my Property Filter as shown.  (Note the Formula).  It's mostly *.asm files, but I think that's because *.par and *.psm almost always have Density and Accuracy in the Custom stream.

[image]

farfilli, 2024-09-10:
You are right Robert if the Custom property is empty the property is not created.
I'm going to investigate, there was an issue about this scenario.

rmcanany, 2024-09-10:
OK.  It's not the end of the world if we can't get it to work.  Seems to me that if someone uses Custom properties, they'll have some in their templates already.  If they don't, it's not an issue.  I guess we could issue an error if we encounter the need to add a property and the Custom property is empty.

farfilli, 2024-09-11:
Fixed with #133