rmcanany/SolidEdgeHousekeeper #127: Plans/Recipies/Presets

rmcanany, 2024-08-22:
Provide a way to load in settings for each step in the workflow.  I currently use a checklist to prep a job for release.  Here's part of it:

```
FIND UNUSED FILES
Select top level assembly and its folder(s).  Bottom up search, report unused files.
    Bottom up search
        Draft files same name as models.
    File list
        Sorted in alphabetical order

Update file list, check for unused files.

UPDATE PART COPIES
Select top level assembly and top level folder(s).  
    Uncheck Report files unrelated to top level assembly
    Bottom up search
        Draft files same name as models.
    File list
        Sorted in dependency order
            Uncheck Include files with no part copies

Update part copies AND Part copies out of date (Part and Sheetmetal).  
Open Save (Assembly), or possibly open the top level assy manually.
```

rmcanany, 2024-10-25:
I'm thinking about doing this next.  I don't think it should be too hard.  My instinct is to use a Dictionary, but want to try the Class approach you suggested.

I've sketched out what I think it might look like below.  Thoughts or examples you may have would be welcome.

In `Form_Main`:
`Public Property Presets as List(Of Preset)`

Somewhere else:
```
Public Class Preset
    Public Property Name As String
    Public Property TaskList As List(Of Task)
    Public Property TaskListJSON As String
    Public Property FormSettings As Dictionary(Of String, Dictionary(Of String, String))
    Public Property FormSettingsJSON As String

    Public Sub New()
    End Sub

    Public Sub ApplyPreset()
    End Sub

    Public Sub Save()
    End Sub

    ' Etc.

End Class
```

farfilli, 2024-10-26:
FormSettings should be a List Of(FormSetting)

FormSetting should be a Class with it's Name and Value properties

rmcanany, 2024-10-27:
Not yet tested thoroughly, but I think it's working.  

The control is currently on the General Tab.  If you have an idea for a better location, please chime in.

One thing to note, if you want a Preset to remember Task selections, that has to be enabled on the General Tab before saving the preset.

farfilli, 2024-10-28:
I always thought to have it on the top of the task list; I did move it with a cup-paste but stopped working so I didn't commit

[image]

rmcanany, 2024-10-28:
I thought it should be visible for all tab pages, so I put it just above the status bar.

[image]

Incidentally, I looked into how to move a control from one parent to another.  With the Form active in VS, go to View > Other Windows > Document Outline.  There, you can drag/drop a control as needed.  Here's a pic.

[image]

farfilli, 2024-10-29:
It seems a good position, I tweaked it a bit the toolbar to show buttons in the correct size.
Please note that if you save a presets with an existing name it asks to overwrite but you will have duplicate elements in the dropdown list. Also a deleted preset reappeared after the save.

rmcanany, 2024-10-29:
OK.  I can fix that.  If you get a chance, please make a PR.  I don't want to create a bunch of merge conflicts.

farfilli, 2024-10-29:
> OK. I can fix that. If you get a chance, please make a PR. I don't want to create a bunch of merge conflicts.

I thought I had done it... it is now 👍

rmcanany, 2024-10-29:
Those two bugs should be fixed now.  It is supposed to work without deleting the Preferences folder, but if it acts up, please give that a try.

farfilli, 2024-10-30:
It seems that when loading a preset nothing happens.
I did delete the preferences folder, but it still doesn't work.

rmcanany, 2024-10-30:
I'll look at it.  If it's a Task selection not working, make sure `Remember selected tasks between sessions` is turned on.

farfilli, 2024-10-30:
Nope, its not task selection but task list and everything else

rmcanany, 2024-10-30:
Sure enough.  That was working before.  I'll track it down.

I'll check the comments in a bit.

rmcanany, 2024-10-30:
I think I found the issue.  Just pushed a commit.  Please let me know if something is still not working.

farfilli, 2024-10-30:
Tasks are still not loaded

I noticed that the windows size and position are saved within the preset, I would prefer it always stays the same and just saves on close and restore on load.

rmcanany, 2024-10-30:
I swear that was working, but you're right.  Made another fix and uploaded just now.  Appreciate all the troubleshooting.

I'll have to look into the repositioning and resizing.

rmcanany, 2025-01-15:
Form position and size no longer saved with the Preset.

farfilli, 2025-01-16:
Thicked tasks are saved? It seems this part is not even saved between sessions

rmcanany, 2025-01-16:
There's a setting `Remember selected tasks between sessions` on the Configuration Tab -- General Page.  Enable that, then save your Preset.

farfilli, 2025-01-16:
Ticks are saved on presets as well?

Ideal workflow is to select a preset that contains only the tasks I want to run and they should be already "ticked"

rmcanany, 2025-01-18:
Closing as complete.

closed 2025-01-18 by rmcanany