rmcanany/SolidEdgeHousekeeper #129: TeamCenter Interoperability

rmcanany, 2024-08-26:
Process using the SEEC API.  Could add a page on the Configuration Tab for any required global setup.

AalianKhan, 2024-08-30:
Hello, I am working for John Mathews and I want to work on this. But it seems like the project is locked in visual studio. Do you or anyone else know how I can unlock the files so I can work on them, I have made my own branch but it still doesn't show the source files when trying to open the project (sln) file
[image]

If someone can help me with this, that will be great.
Thanks

rmcanany, 2024-08-30:
@AalianKhan 
First of all, welcome!  I'm looking forward to seeing your handiwork.  
You probably know a lot more about Git and GitHub than I do.  I looked at the repo on your GitHub page and it appears to have everything I do.  

@farfilli @ChrisClems 
Have either of you ever run into this?

ChrisClems, 2024-09-01:
> @AalianKhan 
> First of all, welcome!  I'm looking forward to seeing your handiwork.  
> You probably know a lot more about Git and GitHub than I do.  I looked at the repo on your GitHub page and it appears to have everything I do.  
> 
> @farfilli @ChrisClems 
> Have either of you ever run into this?

I use Jetbrains so I haven't seen this one before. I have Visual Studio at work so I'll try to see if I can reproduce it after the holiday.

farfilli, 2024-09-01:
I would suggest you to install GitHub Desktop to manage the branches and pulls, pushes and so on

AalianKhan, 2024-09-03:
Thanks guys, 
I will definitely discuss with you the game plan for this. 
But first opening the project is the first hurdle. I haven't run into this ever and there is not much that I can find online. 
I tried installing GitHub desktop and it's the same deal with that. Then I tried JetBrains trial version, and I can open and edit the files, but I can't open the designer for the windows form. 

@ChrisClems Let me know what you find. 
Thanks

ChrisClems, 2024-09-03:
@AalianKhan Have you tried re-cloning the repo? I just did a fresh clone of my fork in Visual Studio and it appears to be working. The red check marks in your solution explorer indicate something changed after cloning.

Try deleting the entire solution from disk. Open up a fresh session of VS. Choose "Clone a repository" and make sure you're using this link: https://github.com/AalianKhan/SolidEdgeHousekeeper.git

Give it time to clone and refresh the status and see what is reported in solution explorer immediately after.

AalianKhan, 2024-09-06:
Hey guys, 
So, my Visual Studio is a bit weird, but I enabled to show all files option, and it is showing everything and I can edit other files as well but still shows the lock icon and such. Hopefully it doesn't cause any issues in the future

I was looking at the code and I was thinking the best way to do this is in configuration add a tab to put Teamcenter login and server information there. Then in the home tab, add another button to open a new form to add team center files. where it uses `GetListOfFilesFromTeamcenterServer` to search files in the server. The user selects which files it wants and once confirms, `DownladDocumentsFromServerWithOptions` downloads the files to cache and use `GetPDMCachePath` to append the path to the name and add that to list view. Then the tasks should be able to play with the files as. Once the execution of all tasks are finished, we can run a check if the file path contains `GetPDMCachePath`, if it does then we can run `CheckInDocumentsToTeamCenterServer` to the documents that have them. 

Let me know any thoughts you have with this.

farfilli, 2024-09-06:
About Visual Studio, did you clone the project and work on your branch? In this way you can edit the files as much as you want and when done you can commit changes. Robert then will review the commit and merge or reject.

rmcanany, 2024-09-06:
That sounds great!  Nice and clean.

You've probably worked this out, but I'm wondering about checking the files back into TeamCenter.  If a user needs to make more than one run, would the files get checked back in after each?  Or what if any errors are reported?

rmcanany, 2024-09-06:
I asked Google about those lock icons.  Here's what it said:

Those are source control status icons.

    Lock means the file is checked in and has no changes.
    Red check mark means the file is checked in, but has pending changes.
    Green plus means this is a new file not yet checked in.

farfilli, 2024-09-06:
To be honest I marginally paid attention to those icons 😂

rmcanany, 2024-09-06:
Oh, sure enough.  I have them too.  I never noticed!

AalianKhan, 2024-09-10:
Ah I see, that makes sense. it is still kind of weird that I have to enable the show all files option. But its working so should be fine. 

I pushed a few commits to my branch to get team center working. I got the Login system down, and now I am working on sending a search query to the server and it is returning some files. For some reason, with my server it doesn't return anything no matter the query. If one of you guys can test if it works with your servers, that will be appreciated.

rmcanany, 2024-09-10:
Hi Aalian,

I'm really a noob when it comes to all things Git.  I went to your Repo and just saw one branch -- master.  Also, it says it's 1 commit and 32 commits behind.  Don't you have to merge our two Repos somehow?  Then, and I'm not sure at all on this, but wouldn't you do a Pull Request so I could merge and download?  One other thing -- I work strictly local, so not sure I can even test what you have done.

@farfilli @ChrisClems would either of you be able to straighten me out on any of this?

farfilli, 2024-09-10:
Before so any change it is always a good practice to synch the branch. 32 commit behind would easily lead to conflict that requires a lot of time to be fixed.

To test another branch you can just download it locally on another folder. GitHub Desktop really helps a lot in doing all this activities and more.

rmcanany, 2024-09-10:
So Aalian would synch, then I would clone from his repo and (I guess) give it a new name?
[image]

farfilli, 2024-09-11:
@rmcanany on the GitHub page of @AalianKhan, you can directly open the code with GitHub Desktop and VisualStudio.
In this way, you can do a quick test on what he's doing without messing with the master\branch
[image]

ChrisClems, 2024-09-11:
> Hi Aalian,
> 
> I'm really a noob when it comes to all things Git. I went to your Repo and just saw one branch -- master. Also, it says it's 1 commit ahead and 32 commits behind. Don't you have to merge our two Repos somehow? Then, and I'm not sure at all on this, but wouldn't you do a Pull Request so I could merge and download? One other thing -- I work strictly local, so not sure I can even test what you have done.
> 
> @farfilli @ChrisClems would either of you be able to straighten me out on any of this?

Correct that @AalianKhan will want to sync up. On the main page of their fork, there is a "Sync fork" dropdown that has an "Update branch" button that will handle it automatically on their end. As long as you're not working on the same code, things should merge without conflicts without keeping everything up to date all the time, but it's still a good idea as the further out of sync you get, the more difficult it will be to merge changes.

As for how to test their branch, there are a lot of ways to skin that cat. @farfilli's way is probably the path of least resistance. You can also just download the repo as a zip, extract it, and open the solution, no git skills needed. I don't know how to do the equivalent in Visual Studio, but in Rider I can just add an additional remote. Then I create a new local branch and merge the branch from the other remote into it. That gives me all the local power of git merging/rebasing using their remote in my IDE.

I highly recommend checking out Oh My Git: https://ohmygit.org/ It's a game that slowly exposes you to git concepts.

farfilli, 2024-09-11:
Ohmygit looks impressive, I will give it a go. Thanks!

AalianKhan, 2024-11-20:
fyi #176

rmcanany, 2024-11-21:
Sorry for my late response. I'll be back in the office next week.

rmcanany, 2025-04-22:
@AalianKhan This is working now.  Thank you for all the hard work!  I credited you and John in the [Readme ](https://github.com/rmcanany/SolidEdgeHousekeeper#readme) and [Release Notes](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/release_notes.md).  Tell your friends (and future employers)!

Closing as complete.

closed 2025-04-22 by rmcanany