rmcanany/SolidEdgeHousekeeper #134: Create a proper Help section

farfilli, 2024-09-13:
At the moment the whole Housekeeper help\guide is written in the Readme file.
There is the need to write the guide more structured with proper sections such as UI, File lists, tasks, external programs, and so on.

The readme file would be back to what it should be, with a brief introduction to the project links to the resources, and more.

Open to consideration and what technology use about it, I think keep using MD file would be good as long as they are accessible through GitHub

rmcanany, 2024-09-13:
The task help sections are actually separate.  The headers and text for each are copied into Readme.md at startup.  That happens in `UtilsDocumentation.BuildReadmeFile()`.  Everything else is in the main file as you point out.  

I found [this link](https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab) which shows how to expand/collapse sections.  That might be a start.  I'll play around with it when I get a chance.

ChrisClems, 2024-09-24:
If we wanted to go all-in on a documentation page, the best option would probably be to set up some automated actions to build and deploy it on Github Pages. It's free. It's a bit of work to get set up initially but after that it should be completely self-building on updates.

It's been quite a while since I've done anything with Pages, but I can dig into how the Housekeeper docs are built currently and see if I can come up with an elegant solution.

rmcanany, 2024-09-24:
I'm not opposed at all, but please don't put in a bunch of work before bringing me up to speed.  

It's pretty automated right now, except for writing the documentation of course.  From what @farfilli mentioned, I gathered it's more about the navigation.  I agree that could be greatly improved.  My idea for that was to create collapsible sections with the html <details> tag.

rmcanany, 2024-09-29:
I took a stab at this, adding a table of contents for the major topics.  More complex topics, such as Property Filter, have their own supplemental ToC.

If you get a chance, take a look and let me know what you think.

rmcanany, 2024-10-06:
Just ran across this, to me great, format: https://raehart4.gitbook.io/scc-program-terms.   What do you all think?  I can do some research in the coming days.  Come to think of it, is that what you were proposing @ChrisClems?

ChrisClems, 2024-10-08:
> Just ran across this, to me great, format: https://raehart4.gitbook.io/scc-program-terms. What do you all think? I can do some research in the coming days. Come to think of it, is that what you were proposing @ChrisClems?
> 
> Oh, that's not GitHub. It's a separate company. I think I'll keep looking.

That looks like a managed service version of what I was proposing.

This is a super quick overview of the kind of docs toolchain I've been wanting to deploy to my projects: https://dotnet.github.io/docfx/

Once you've got all the up front work done it's basically automatic doc generation on every build in a nice format like you linked.

rmcanany, 2024-10-18:
I might come back to this, but in my opinion the Table of Contents as sufficient to address the navigation issue for now.  Closing as complete.

closed 2024-10-18 by rmcanany