rmcanany/SolidEdgeHousekeeper #137: Refactor default template usage selection to task settings

ChrisClems, 2024-09-25:
I started to type this up in the beta tester thread but figured it would be more appropriate to raise it here.

It sounds like Robin was missing that they had to press the Copy button to propagate the material template set in the config to the relevant tasks. I also missed this at first glance when trying to recreate the issue.

I think it would be more inherently intuitive to move this to auto-update via a checkbox in the task.

That is to say, under "Check material not in material table" task settings, above the Matl Table text box, there would be a checkbox to specify that the user would like to use the material table specified in the Config/Templates tab. When checked, the material table for the task would updates from the path specified in the templates, and again on every `TextBoxMaterialTable_TextChanged` event.

I know this is a bit more complicated to maintain as it must be added to each task, but I think it will be more intuitive for the users.

Thoughts?

rmcanany, 2024-09-25:
Yeah, that's a good idea.  I was thinking about a warning dialog on the templates page, but I like your solution more.  There are only like three commands that use templates.  It won't take long.  I'm on it!  Thank you!

rmcanany, 2024-09-26:
It should be working now.  If you get a chance to try it, please let me know if you run into any issues.

ChrisClems, 2024-09-26:
it's working on my end and looks slick.

closed 2024-09-26 by ChrisClems