rmcanany/SolidEdgeHousekeeper #139: Consider showing properties on the file list

rmcanany, 2024-10-02:
Suggested by @Careframe1 on the SE Forum:
https://community.sw.siemens.com/s/question/0D5Vb00000Lej9IKAR/solid-edge-housekeeper-v20243-is-released

Also be able to sort on property values.

farfilli, 2024-10-03:
I considered doing that some time ago, but then I got busy and didn't try it.

For sure don't even try to use the Solid Edge property APIs, it would have a bad performance impact on a large list of files.
I suggest using the same method as in the direct properties editor; it should be fast enough.

I can take this task.

rmcanany, 2024-10-03:
OK, I assigned it to you.  No rush, obviously.

rmcanany, 2024-10-07:
I'm thinking about changing ListOfColumns to store the whole property formula (eg %{System.Title}) instead of just the name (Title).  

Two reasons:
1.  We could reuse UtilsCommon.GetOLEPropValue(), eliminating UtilsFileList.FindProp.
2.  We could provide the option to use DM if someone wanted Material or other properties not in structured storage, and "|R1" props for Draft files.

The second option would incur a performance hit, but the user could decide if it's worth it.

I can make the change.  I'm mostly wondering if you see anything I'm missing.  (BTW, I would strip the prop name from the formula for the column header text.)

farfilli, 2024-10-07:
Yes Robert sounds good. I would store it in a duupe so we can also store the visibility.
It would be a list(of String, Boolean)

rmcanany, 2024-10-07:
Cool.  I'll play around with it today.

farfilli, 2024-10-08:
Partially completed with #143 

A possible future improvement could be to store column width and ability to reorder columns

farfilli, 2024-10-08:
@rmcanany On the Column Properties localized ones don't work, I have to add manually the English name. Can you take a look?

rmcanany, 2024-10-08:
@farfilli I will take care of that.  

I am going to use UtilsCommon.SubstitutePropertyFormula, which already does it.  I'm also going to add a Formula property to the PropertyColumn class.  It would be in the correct format, eg %{System.Title}.

I like the magic delete buttons, by the way.  That slick how you did that.

On a related note, what does this do?
`        Me.CLB_Properties.Items.AddRange(New Object() {"Name", "Path"})`

farfilli, 2024-10-08:
It had the initial two properties name and path to the checkboxlist

rmcanany, 2024-10-08:
Oh, got it.

farfilli, 2024-10-08:
But they start unchecked, perhaps add them programmatically at first startup or if they miss in the configuration file. It needs to be initialized in some way.

About the magic button its just a regular button that show/hide based on mouse position

rmcanany, 2024-10-08:
OK, I'll make sure they're initialized correctly.  That's no problem.

rmcanany, 2024-10-08:
@farfilli 
I added Formula to PropertyColumn and switched from using FindProp to SubstitutePropertyFormula.  When you get a chance, could you test localized properties?

I haven't done the saved settings yet.  That's next.

farfilli, 2024-10-09:
Localized properties still doesn't work, in the image you can see the localized one beside the english one

[image]

rmcanany, 2024-10-09:
I added an initializer for template properties.  It's up on GitHub.  Not sure that's the whole story.  Let me know if it works.  

While you're at it, it would be good to check Edit Properties, too.  It uses the same code and might tell me if something is broken in there.

farfilli, 2024-10-09:
The ListOfColumns List doesn't contain "Name" and "Path" anymore; they are mandatory to be in the list at positions 0 and 1.
It seems they are missing now therefore it is not possible to show/hide them.
The column configuration is not maintained from one run to another

Localization still not working
[image]

farfilli, 2024-10-09:
Fixed Path and Name startup with #144

rmcanany, 2024-10-09:
Thanks for that.  I'm still working on the saved settings part.

rmcanany, 2024-10-09:
Can do.  To save me some digging, do you know off the top of your head how to get the widths?

rmcanany, 2024-10-09:
Oh sorry, I meant get/set on the `ListView` object.  I found it: `ListViewFiles.Columns.Item(i).Width = W`

rmcanany, 2024-10-09:
I think the saved settings are working now.  I haven't added the column width piece yet.

farfilli, 2024-10-10:
It seems to work, good job

farfilli, 2024-10-10:
I have implemented the column width property with #145 

I did try to implement the width saving but as I understand if I update the JSON list then an update is fired that goes into a loop.
I blocked the loop initially by a boolean property, then a better method was to compare width values and if they don't really change skip the update.

farfilli, 2024-10-10:
Yes I'm done but don't forget that localized properties still doesn't work

rmcanany, 2024-10-10:
Oh, I missed the part about localized properties.  Did you say they're working on Edit Properties?

rmcanany, 2024-10-10:
Never mind.  I found the issue.  Now to fix it...

rmcanany, 2024-10-10:
I don't know if it's working, but at least it is compiling.  

If it works in the list properties code, it should work everywhere.  If you want to double-check, here are the other places SubstitutePropertyFormula is called.
- TaskEditProperties: FindText and ReplaceText
- SaveDrawingAs: SubdirectoryFormula
- SaveModelAs: SubdirectoryFormula

rmcanany, 2024-10-10:
What do you think about Export List including the extra columns?

farfilli, 2024-10-11:
> I don't know if it's working, but at least it is compiling.
> 
> If it works in the list properties code, it should work everywhere. If you want to double-check, here are the other places SubstitutePropertyFormula is called.
> 
> * TaskEditProperties: FindText and ReplaceText
> * SaveDrawingAs: SubdirectoryFormula
> * SaveModelAs: SubdirectoryFormula
> 
> **Edit**: The SaveAs commands don't use structured storage, so they should be working already. Hopefully.

I have tested localized properties in Direct Edit and Solid Edge Edit on FindText, and ReplaceText both in PT and in EX and it seems to work fine.

Tested SaveModelAs in SubdirectoryFormula and it works ok

There is a bug with Custom properties that seems to be case-sensitive:

These settings work fine:
[image]

These settings lead to an error:
[image]
[image]

Please note that in the file the custom properties is written in upper-case "OGGETTO 1"

farfilli, 2024-10-11:
I did try to implement the column update after the property edit but in the case of Solid Edge edit I got a cross-thread error; not a problem in direct edit. I did try to pass the listviewitem as a parameter in the Process sub but it wasn't a good idea.

I think the best approach would be the thread passing back info to update the listviewitem values after each execution, it may be something to insert in the dictionary of errors, which would be a dictionary of execution or something like that.

farfilli, 2024-10-11:
@rmcanany I have implemented property edit from FileListView, just double-click a property to edit it.
I have created a new method in TaskEditProperties to edit a single file property

rmcanany, 2024-10-11:
Good deal.  I like the idea of editing the properties like that, but I got an exception.  The file isn't open in another application as far as I know.

[image]

Not sure how to handle the text case issue.  It's bombing out because `TemplatePropertyDict` does not have that key.  I guess I could convert everything to upper case.  I would have to see what other problems that might create.  

Is something like that going on in `UC.GetOLEPropValue()`?  Could you explain what `Function(Proper) Proper.PropertyName` does?  Also why sometimes upper case and other times lower case?

`OLEProp = System_Properties.Properties.First(Function(Proper) Proper.PropertyName = "PIDSI_" & PropertyName.ToUpper)`
`OLEProp = System_Properties.Properties.FirstOrDefault(Function(Proper) Proper.PropertyName.ToLower Like "*" & PropertyName.ToLower & "*")`
`OLEProp = Custom_Properties.UserDefinedProperties.Properties.FirstOrDefault(Function(Proper) Proper.PropertyName = PropertyName)`

I've run across the cross-thread issue before.  A programmer I know said that's what `STDOUT` and friends are for.  Once I get it working for error reporting, I can look at updating the ListView.

farfilli, 2024-10-11:
I don't know what may lock the file, I did test with some files in a folder and was working correctly. Didn't test it extensively.

About the ToUpper is mandatory in streams cause they are written that way. The ToLower is just an easy way to compare it could be ToUpper as well.

About the custom properties being case sensitive I didn't investigate the reason,

farfilli, 2024-10-11:
That Function is to get the property by name without the need to parse all the properties stream.

rmcanany, 2024-10-11:
Got it.  

Changing a property in the ListView works for me sometimes and not others.  I'll see if I can narrow it down.

Is there a reason for `UpdateSingleProperty` to be in `TaskEditProperties`?  I'm thinking about moving it to  `UtilsCommon`.

farfilli, 2024-10-11:
I didn't know how the localization worked and thought it had to be there. Move it where you think is it's place 👍🏻

farfilli, 2024-10-12:
Need to implement the addition of custom properties in case they are missing

rmcanany, 2024-10-12:
Forgot about that one.  I can play around with it.

On a related note, I downloaded SSView to get a better idea about structured storage.  Poking around, I noticed that Material, Material Thickness, and Sheet Metal Gage are all stored in there.  For some reason I thought they weren't.  Was the issue with OpenMcdf not being able to access the correct stream or something?

farfilli, 2024-10-12:
I haven't found the density, that's the main concern about changing material.

All other things you found are already managed by direct editing.

I can do it on Monday, It was just a reminder 😅

rmcanany, 2024-10-12:
Oh, I didn't know that direct edit had access to all those.  Great!  

Do we need to update these lists, or is it handled some other way?
```
        Dim SIList As New List(Of String)
        SIList.AddRange({"Title", "Subject", "Author", "Keywords", "Comments"})

        Dim DSIList As New List(Of String)
        DSIList.AddRange({"Category", "Company", "Manager"})

        Dim FunnyList As New List(Of String)
        FunnyList.AddRange({"Document Number", "Revision", "Project Name"})
```

I found Density with SSView:
[image]

I would be glad to let you handle the OpenMcdf stuff.  I reread your Issue on their GitHub page.  I could maybe figure it out, but not in this lifetime.

farfilli, 2024-10-12:
I Will take a look at that, it opens to more capabilities 👍🏻

rmcanany, 2024-10-12:
Sounds good.  

I'm still getting random exceptions when editing the ListView and now when doing Edit Properties outside of SE.  For Edit Properties it's random.  Run it two times in a row and different files bomb out.  It's usually 3 or 4 files out of a hundred.  Can you think of something I could have messed up on my end?

farfilli, 2024-10-12:
Did you try with the file in the training folder?

rmcanany, 2024-10-12:
Same story in the training folder.  That's a good idea about getting the latest build.  How do I do that?

rmcanany, 2024-10-12:
I downloaded the zip files.  Do I just place the dlls in My Project folder?

rmcanany, 2024-10-12:
Come to think of it, it's happening when the file stream is being created -- before OpenMcdf even gets its hands on it.  I found a bunch of info on the issue [here](https://stackoverflow.com/questions/26741191/ioexception-the-process-cannot-access-the-file-file-path-because-it-is-being).  I might try a few of those suggestions.

rmcanany, 2024-10-12:
I got to work 2 times in a row by adding these lines in two locations in UtilsFileList.  It seems to have fixed ListView edit as well as Edit Properties.
```
cf.Close()
fs.Close()
cf = Nothing
fs = Nothing
```

farfilli, 2024-10-13:
> I downloaded the zip files.  Do I just place the dlls in My Project folder?  Also, net40 or netstandard2.0?

I usually download the latest code, build the DLLs and copy them in the project folder (overwriting the current ones)

should be net40

farfilli, 2024-10-14:
> Need to implement the addition of custom properties in case they are missing

I have implemented it by 8a6320c205f31162e8532c22e8a087ce801f9d61

**Please note that there is an important code change**

farfilli, 2024-10-14:
> Oh, I didn't know that direct edit had access to all those. Great!
> 
> Do we need to update these lists, or is it handled some other way?
> 
> ```
>         Dim SIList As New List(Of String)
>         SIList.AddRange({"Title", "Subject", "Author", "Keywords", "Comments"})
> 
>         Dim DSIList As New List(Of String)
>         DSIList.AddRange({"Category", "Company", "Manager"})
> 
>         Dim FunnyList As New List(Of String)
>         FunnyList.AddRange({"Document Number", "Revision", "Project Name"})
> ```
> 
> I found Density with SSView: ![density_in_structured_storage](https://private-user-images.githubusercontent.com/64286960/375971662-4ba7f0ed-4755-4cb5-96d2-941e3ab4d45e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg4ODY4OTgsIm5iZiI6MTcyODg4NjU5OCwicGF0aCI6Ii82NDI4Njk2MC8zNzU5NzE2NjItNGJhN2YwZWQtNDc1NS00Y2I1LTk2ZDItOTQxZTNhYjRkNDVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE0VDA2MTYzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlYTA0MDQ1MTQzZWUyMDVhNGRkNTRjOTZlZTIwMjczNTI4N2VkYmMwNDdhZTYxZThhMjZkMGYyOGFjZmI1NGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ToylyMP2bq2SRIhRojWcGRNVHU-lxWtAfTi-Pk-Pves)
> 
> I would be glad to let you handle the OpenMcdf stuff. I reread your Issue on their GitHub page. I could maybe figure it out, but not in this lifetime.

The density and Accuracy shown in the picture are present in the custom list as exposed but can be deleted.
I noticed that they are recreated if you don't have them while changing the material through the Solid Edge interface
If you change the value of those properties instead the Material is not updated, in my opinion, those properties should be read-only.

Changing the density through SSView doesn't work, I didn't try with OpenMCDF, in any case, it would not update the Material density.
Changing the name of the material will not update the other physical properties therefore a method to read the mtl file and set the properties would be required, and in the end, the Density will not be updated cause I haven't found it in the material section.
[image]

We could instead implement perhaps those properties as read-only for reference purposes, let me know what do you think

rmcanany, 2024-10-14:
OK.  That makes sense -- those are exposed variables, so probably don't propagate back to the variable table.  I like the read-only idea.

For the release I'm working on, do you think we're ready to include the changes to the ListView?  If not, I think I can just disable the button on the toolbar.  Let me know.

rmcanany, 2024-10-14:
By the way, I checked one file where I had changed a custom property and it showed up correctly in the Windows property page.

farfilli, 2024-10-14:
Did you change it before the last update of the DLL ?

rmcanany, 2024-10-14:
Agreed.  I'm going to disable the button for now.

I didn't update the DLL on my end.  Did you update it on your side?

farfilli, 2024-10-14:
I did update it on 09/09/2024, can you check if your edit occurred previously?

We need to create a list of checks to do when updating these DLLs cause they are evolving fast.
When the next release will be on NuGet we will move on it and consider it not experimental anymore.

rmcanany, 2024-10-14:
My edit was in the last couple of days.  Maybe it has something to do with localized properties?

farfilli, 2024-10-14:
> My edit was in the last couple of days. Maybe it has something to do with localized properties?

Custom properties shouldn't be affected by localization, does the code try with them too?

rmcanany, 2024-10-14:
OK, so that's not in the released version, right?

rmcanany, 2024-10-14:
Oh, right about custom properties.  For those it still uses the lookup table, but the English name is the same as the custom property name.

rmcanany, 2024-10-14:
I got the click event to fire by changing `HeaderStyle` to Clickable.  

If you have time, you're welcome to do the sorting.  Otherwise I think I can do it with [this sample code](https://learn.microsoft.com/en-us/dotnet/api/system.windows.forms.listview.columnclick?view=windowsdesktop-8.0).

[49 more replies]

closed 2024-11-29 by rmcanany