rmcanany/SolidEdgeHousekeeper #140: Change Document Status UI

rmcanany, 2024-10-02:
Rather than a matrix of radio buttons, a list something like:

FROM            TO
Available        [ComboBox]
In Work          [ComboBox]
...

Create Properties to store the values, like
Public Property StatusFromAvailableTo as String
Public Property StatusFromInWorkTo as String
...

Hide controls that don't currently apply, rather than just disabling them.

farfilli, 2024-10-03:
I'm not a fan of ComboBox, they need double clicks and don't show the options you got.
I vote for a well designed matrix

rmcanany, 2024-10-03:
Appreciate the input.  I'll give some more thought.

rmcanany, 2024-10-18:
Closing as not urgent.  Hoping Francesco gets tired of all the gray and fixes for me! :smile:

farfilli, 2024-10-21:
What's the point of closing it as not urgent?
If it's not urgent doesn't mean it's closed, it should stay open until its completed or closed as rejected.

Otherwise, how do you track what to do and what not?

rmcanany, 2024-10-21:
Good point.  Got a little over-tidy on my list cleanup.

closed 2025-04-15 by rmcanany