rmcanany/SolidEdgeHousekeeper #155: Search in not authorized folders throw an exception

farfilli, 2024-10-18:
I have added my document folder as a source with subfolders search.
It happens that for some reason the image folder is not accessible.

The code should be smart enough to handle this situation

[image]

rmcanany, 2024-10-18:
Agreed.  I took a quick look at `FileIO.FileSystem.GetFiles` and didn't see an option to skip read-only directories.  I guess we could add a try/catch and report the error given.  At least it wouldn't crash that way.

farfilli, 2024-10-21:
First search for all the subfolders then for each subfolder search files, in this way it's possible to handle single subfolder error.
Done in 495d90e6c0f60bbfa69c182b9b549560075a9257

rmcanany, 2024-10-21:
Good idea.  Tell me if I'm wrong, but shouldn't `SearchAllSubDirectories` be `SearchTopLevelOnly`?
```
                        For Each tmpFolder In tmpFolders

                            Try

                                tmpFoundFiles.AddRange(FileIO.FileSystem.GetFiles(tmpFolder,
                                FileIO.SearchOption.SearchAllSubDirectories,
                                ActiveFileExtensionsList.ToArray))

                            Catch ex As Exception

                                Dim s As String = "An error occurred searching for files.  Please rectify the error and try again."
                                s = String.Format("{0}{1}{2}", s, vbCrLf, ex.ToString)
                                MsgBox(s, vbOKOnly)

                            End Try

                        Next
```

farfilli, 2024-10-21:
You are right, I didn't pay attention to that part of the code. You do the fix?

rmcanany, 2024-10-21:
I'll fix it.  Good point about the message box.  I'll move it outside the loop.

rmcanany, 2024-10-21:
Fixed.  One question -- is a "TXT error message" the same as a MsgBox?

Incidentally, this was only finding the first level of subs.
`Dim tmpFolders As String() = Directory.GetDirectories(Source.Name)`

I changed it to this.
`Dim tmpFolders As String() = Directory.GetDirectories(Source.Name, "*", SearchOption.AllDirectories)`

farfilli, 2024-10-22:
It's how you show errors at the end of files processing.

Good catch on the directory retrieval 👍🏻

farfilli, 2024-10-22:
Unfortunately the SearchOption.AllDirectories works in the same way as files
[image]

rmcanany, 2024-10-22:
Looks good.  

I see my error reporting is still not ideal.  It should wait until all files are processed before providing feedback.  

I'm going to leave it for now and add a note in the error-handling issue to take care of this when I do that.

rmcanany, 2024-10-24:
Closing as complete.  Thank you!