rmcanany/SolidEdgeHousekeeper #158: Show FOA members separately on the list

rmcanany, 2024-10-21:
Maybe add an option to show or not.

farfilli, 2024-10-21:
My idea is to show a FOA like this:
[image]

rmcanany, 2024-10-21:
Nice and clean.  As usual!

farfilli, 2024-10-21:
> Maybe add an option to show or not.

I agree, checking if it's a FOA would have a performance impact.
A method could be to use openMCDF to check if a stream named "Master" is present.

farfilli, 2024-10-22:
With commit bc17d0a55abc0c19e834c124073ec152d6bbe4e4 I have added initial support for FOA properties, at the moment only Master properties are handled.
I haven't checked if the test has an impact on the file list update and on the Task Edit properties

rmcanany, 2025-01-18:
Thinking through this a bit and wondering if it's necessary.  

At least for `TaskEditProperties`, presumably we would want each FOA member to be handled differently in some way.  I can't think of a clean way running in batch mode to accomplish that.

For other tasks, where we want to do the same thing to each member, like checking interference, we could update the task itself to cycle through each member in turn without bothering the user with additional information.

rmcanany, 2025-02-17:
I decided this would be best handled in applicable Tasks.  There are just a handful of them and  SaveModelAs is already done.  Rather than updating proactively, I'm going to wait until there is a request from a user.

Closing as not needed.