rmcanany/SolidEdgeHousekeeper #164: Problem on handling properties

farfilli, 2024-10-25:
I realized that the whole handling of properties is written considering them always to be strings.
Actually, Solid Edge supports four types of properties: Text, Number, Yes\No, and Date.

Despite the fact that most of the users only use the text type, the code should be strong enough to recognize and handle all the situations.

The problem is more critical with the OLE methods cause you can't write a decimal value into an integer property; yes, despite the use choose numeric, internal integer and decimal values are handled with two different VTType, VT_I4 for integers and VT_R8 for decimals.

I will start to fix the OLE way soon, I just need a feedback from the openmcdf project

rmcanany, 2024-10-25:
When setting a non-OLE property's value, I currently enclose it in a Try/Catch block.  If it bombs out, the error says the command only works on text type properties.  That could be improved.  I think the only place it's used is in Edit Properties.

One exception to the text-only rule is Status changing.  There I cast the Status and Security constants to integer before assigning.

farfilli, 2024-10-25:
What if an user have numeric and date properties? And what about the Hardware property that is a yes no ?
I would try to implement all the types, of course is not an high priority issue.
I will work on the OLE side.

rmcanany, 2024-10-25:
Good points.  Added them to https://github.com/rmcanany/SolidEdgeHousekeeper/issues/156

rmcanany, 2024-10-31:
Running into a snag getting the property type.  Couldn't find anything in the `SolidEdgeFileProperties.Property` API.  I tried this.
```
  Prop = CType(PropertySet.Item(i), SolidEdgeFileProperties.Property)
  PropName = Prop.Name
  Dim TypeName = Microsoft.VisualBasic.Information.TypeName(Prop)
  Dim destType As Type = SolidEdgeCommunity.Runtime.InteropServices.ComObject.GetType(Prop)
```
Using VB, `TypeName` = `"IJProperty"`
Using SECommunity, `destType` = `SolidEdgeFileProperties.Property`

Ideas?

rmcanany, 2024-10-31:
Never mind.  I just needed to check `Prop.Value` instead of `Prop`.

Interestingly, the variable type `Double` did not appear.  Things like Poisson Ratio were listed as `String`.

The four types I saw were: `String`, `Integer`, `Boolean`, and `Date`

farfilli, 2024-11-05:
Implemented SetOLEPropValue in ProcessInternal handling of different types properties through Compound Files is completed

rmcanany, 2024-11-05:
OK, cool.  I won't get a chance to test it right away.  We're heading out of town tomorrow for a couple of weeks.  I'm racing to get the new property filters class integrated before then.

farfilli, 2024-11-07:
Implemented the substitute property formula in SetOLEPropValue
[image]

rmcanany, 2025-01-15:
Except for the issue with `Number` variables on the SE side (https://github.com/rmcanany/SolidEdgeHousekeeper/issues/188) this should be done.

rmcanany, 2025-01-18:
Closing as complete.  Tracking the `Number` property type issue here: https://github.com/rmcanany/SolidEdgeHousekeeper/issues/188

closed 2025-01-18 by rmcanany