rmcanany/SolidEdgeHousekeeper #170: Extend support to more file type

farfilli, 2024-11-05:
This is an important change in the application and needs to be planned with several methods and functions to be updated.
The idea is to extend support for more file types, even if they are not directly related to Solid Edge.

I would implement PDFs first; let's start with some basic tasks like printing and watermarking.
The next batch would be to implement Word and Excel files and start with tasks like save as and print.
Another batch would be to manage step files, parasolids, and so on, and implement the conversion to Solid Edge files.

@rmcanany what do you think? Is it something you would consider or strictly Solid Edge files?

rmcanany, 2024-11-05:
That's a cool idea.  We can talk more when I get back.

farfilli, 2024-11-26:
I have created a discussion about this topic where we can start to plan changes to make it possible and then open specific issues for each one.

closed 2024-11-26 by farfilli