rmcanany/SolidEdgeHousekeeper #174: Save part as use a diffrent name than the Partname itself

Jojo15702, 2024-11-13:
Hey I would like to save numerous parts in one go as a step-file. But i dont want them named the same name as the .prt-Name itself.

For Example my .prt-Name is "119989-1169.___Gear 1" with "119989-1169" beeing the Partnumber and "Gear 1" the title.

I cant find a Way to save the Step-file with only the Partnumber and not the .prt-Name.
It would be nice if the option is given to Rename the Saved Files according to propertys.

farfilli, 2024-11-13:
I think the property formula %{System.Title} should work, but I will check it and give a feedback tomorrow

rmcanany, 2024-11-25:
Thank you for the suggestion.  I'll take a look.  

In the meantime, you could check https://github.com/rmcanany/SolidEdgeHousekeeper/issues/83 for a workaround.

Jojo15702, 2024-11-27:
Thank you
Unfortunatly the workaround doesn't solve my problem.
I'll keep changing the names manualy. After all its paid time.

rmcanany, 2024-11-27:
Oh, I see what you mean.  The workaround would create another *.par file, not *.stp.  

The filename fix is next on the list.  It won't be released right away, though.  We've got a few other changes in the works before that can happen.

If you like, once I get it working, I could drop a compiled *.exe out to my Google Drive for you to try out.  Let me know.

Jojo15702, 2024-11-28:
Sounds great, let me know when it's ready. Thanks!

rmcanany, 2024-11-29:
Hi @Jojo15702 

Please email me and I'll reply with a link to the `*.zip` file.
[image]

rmcanany, 2024-12-10:
Thank you for the suggestion!  Closing as complete.

closed 2024-12-10 by rmcanany

ih0nza, 2025-01-09:
Hello, I have similar thing: For documents, we use unique file names, and the document number is filled in the file properties as an attribute. For batch exports to DWG, PDF, and STEP, we currently use the Batch.exe program provided with Solid Edge, which we have customized to create a file names like "(DocumentNumber) (Filename).pdf" during export. Is it possible to define in Housekeeper that this "DocumentNumber" is added before the file name? Thank you.

[image]

rmcanany, 2025-01-09:
Hi @ih0nza 
[image]
That preview is still out on my Google drive.  Please email me and I'll send you a link.

ih0nza, 2025-01-10:
Hello thank you for your reply. I need name of exported file like "[document number] [file name].pdf/dwg is there possible to set this? Thank you Odesláno z iPhonu9. 1.

Hi @ih0nza
image.png (view on web)

rmcanany, 2025-01-10:
Yes, you can.  Not with the current version, but instead with the preview version I mentioned.  Here's how you would set it up.  
[image]

rmcanany, 2025-01-10:
Sent the link.  Please let me know how it goes.

rmcanany, 2025-01-10:
One more thing.  Please see the [Templates Page](https://github.com/rmcanany/SolidEdgeHousekeeper#templates-page) of the Readme.  It shows how to retrieve properties from your templates.

ih0nza, 2025-01-11:
Hey!!! Thank you VERY MUCH. There is everything I needed. Exported file
name works and there also is possible to show "Document Number" column in
Home window.

One small thing (not necessary now). When I have seen possibility to save
preset in Tasks I thought that I can save which tasks I have checked,
unfortunately it is only about tasks settings (not bad) but I could welcome
preset for checked tasks. Because I have plan to have about 15 printing
tasks. Better to check 15 checks than print drawings manually, but when I
can save that 15 checks checked It would be nice :)

Thank you once again. You Rock :)
h0nza

rmcanany, 2025-01-11:
Glad it worked!

Regarding the Presets remembering selected Tasks, it should do that.  On the Configuration Tab -- General Page, try enabling `Remember selected tasks between sessions`.  You'll have to re-save the Preset for that option to be captured.

I say "should do that" because it's a new feature and hasn't been thoroughly tested.  It could easily have some bugs.  I would appreciate any feedback.  If you like, I'd be glad to add you to the list of Beta testers.  That's not worth any money, but it ups your "cred" on GitHub.

ih0nza, 2025-01-11:
Yes I noticed that opnion to "remember selected tasks between session" but
it is not exactly what I need, because, sometimes I need just export > 2 or
3 chesks, and then I need only print > 15 checks, and sometimes I want to
make it all together > 17 checks. Time by time I need to switch it. There
should be time when I want to print and couple hours later I need to
export. So there would be nice to have that checks remembered.

About that beta testing I would be glad to test some new features for you
if you want. I am new to GitHub so I dont know about that "cred".

Today I just tested it at home on 2 files if there is what I needed. Later
I will test it on more files (50-100) at work.

Thak you for all.

rmcanany, 2025-01-11:
Not sure I follow about the Presets.  If you could show me some pictures of the various configurations, I may be able to do something.  Chances are, if it would help you, it would probably help others, too.

Yeah, "cred" stands for "credibility".  It's not written down anywhere, or specific to GitHub.  It's just kind of a "cool factor" I guess.

ih0nza, 2025-01-20:
Hello,

When I talked about task pressets: 

I yusually want to do 3 separate actions. 1) Export 2D, 2) Export 3D, 3) Print. When I want to do exports it is easy to select 1 or 2 tasks. But when I want to print I have to select 15 tasks, and when I want to export after print I have to unselect that 15 tasks and then select 1 or 2.

 

1.	Export 2D

 

2.	Export 3D

 

3.	Print

For every paper size I have configured text and position of watermark (1,2,ARCHIV) so thats why I have too many print tasks. Until now we just printed 3 copies of every drawing and then make hand stapms (1,2,ARCHIV) on every paper. I was exhausting so with help from Housekeeper I can make that little automated 😊

 

So when I can save/load selected task it should be easier .

 

Thank you

Best regards

h0nza

rmcanany, 2025-01-20:
You have good timing.  I'm testing Presets right now.

I think it should do what you want.  The trick is to first set up your Tasks like you want.  Before you save the Preset, go to the **Configuration Tab -- General Page** and enable `Remember selected tasks between sessions`.  I just tried it and it is working for me.  It not only remembered the tasks, but also all of their settings.

Note that for the Preset to be loaded, you have to hit the `Load` button.  Also, I just fixed an issue where property filters were not being saved with the Preset.  I'm currently testing that fix.

[image]

[image]

ih0nza, 2025-01-22:
Yes!! It works :) You are great.
now there is everything I needed. When I tried to print couple drawings
(with my 15 print task) there was any error with 2-3 drawings. I
printed them additionally again with "process tasks on selected files". I
will check it again next time and possibly send an error log.

When I was thinking about Homekeeper I had one idea. When we used Batch.exe
(from Solid Edge) for batch export drawings, there was an option to open
drawings "inactive". I usually used it because opening of drawings was much
quicker. Only idea to think about :)

Best regards
h0nza

rmcanany, 2025-01-22:
Good deal.  Yeah, if you get an error again, please open a new Issue and include the error message.  That really helps track things down.

I appreciate your suggestion to provide an option to open inactive.  That's a good idea.  Thank you!