rmcanany/SolidEdgeHousekeeper #175: Restart based on Edge.exe handle count instead of number of parts processed

ChrisClems, 2024-11-14:
I'm working on a script that is outside of my usual wheelhouse. Instead of processing a several hundred part assembly, I'm churning through an entire rev control library of SE files that's in the many tens of thousands of parts. Originally, I did something very similar to HouseKeeper in monitoring my processed file count then restarting the SE app.

I believe what I've found is that Solid Edge is spinning up threads as files are opened and operated on, then leaking the handles to those threads when they are killed. As time goes on, I see thread handles to non-existent processes are steadily increasing.

I implemented a simple monitoring class that watches the handles used by the Solid Edge process. It spins up a thread which, at specified intervals, retrieves the handle count and stores them in a thread-safe variable. Then the handle usage can be checked with minimal blocking while processing parts. So far this is working great as it will only kick in when Solid Edge is actually misbehaving by leaking handles. If whatever you're doing with SE isn't causing the issue, it won't trigger the restarts.

Is this something you'd be interested in implementing as an advanced optional alternative to restarting every X parts processed? The code to actually make it work is surprisingly simple. I'm running checks at 30 second intervals and restarting after 5,000 handles are in use.

Usage looks like this:

```c#
HandleMonitor.StartMonitoring("Edge", 30);
foreach (var file in solidEdgeFiles)
    {
        if (HandleMonitor.GetHandleCount() > 5000)
        {
            // Restart Solid Edge application
        }
        // Do stuff with files
    }
HandleMonitor.StopMonitoring();
```

```c#
public static class HandleMonitor
{
    private static volatile int handleCount = -1;
    private static bool keepRunning = true;
    private static Thread? monitorThread = null;

    public static void StartMonitoring(string processName, int intervalInSeconds)
    {
        if (monitorThread != null && monitorThread.IsAlive)
        {
            throw new InvalidOperationException("Monitoring is already in progress.");
        }

        keepRunning = true;
        monitorThread = new Thread(() => MonitorHandleCount(processName, intervalInSeconds));
        monitorThread.Start();
    }

    public static void StopMonitoring()
    {
        if (monitorThread != null && monitorThread.IsAlive)
        {
            keepRunning = false;
            monitorThread.Join();
            monitorThread = null;
        }
    }

    public static int GetHandleCount()
    {
        return handleCount;
    }

    private static void MonitorHandleCount(string processName, int intervalInSeconds)
    {
        while (keepRunning)
        {
            Process[] processes = Process.GetProcessesByName(processName);
            if (processes.Length > 0)
            {
                try
                {
                    handleCount = processes[0].HandleCount;
                }
                catch (Exception ex)
                {
                    Console.WriteLine($"Error retrieving handle count: {ex.Message}");
                }
            }
            else
            {
                handleCount = -1;
            }

            Thread.Sleep(intervalInSeconds * 1000);
        }
    }
}
```

rmcanany, 2024-11-21:
Sorry for my late response. I'll be back in the office next week.

rmcanany, 2024-11-25:
That's slick!  

I actually turned off the automatic restarting a while back.  The `System.Threading.ApartmentState.STA` in `IsolatedTaskProxy` seems to have fixed my `COM` errors.  However, I'll keep it in mind if it starts misbehaving again.

Thank you!

rmcanany, 2025-01-18:
Copied to Discussions page.  Closing here.

closed 2025-01-18 by rmcanany