rmcanany/SolidEdgeHousekeeper #178: Avoid DesignManager usage

farfilli, 2024-11-21:
Some users have reported errors when using SEHK with older SE versions (ST9).
The error triggers during DesignManager initialization. From an API point of view, I believe that version is still called RevisionManager.

I noticed that the DesignManager is always initialized even when it's not needed at [this ](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/b39b5c311e917a2dc0b6cf6d3088338bf522d264/My%20Project/UtilsExecute.vb#L319)line of code.

I see that DesignManager is used to change file status if this function is enabled. In my opinion, openMCDF would do the job fast and without the need for DesignManager; in this way, we can guarantee RevisionManager compatibility.

rmcanany, 2024-11-25:
Agreed.  See https://github.com/rmcanany/SolidEdgeHousekeeper/issues/179

rmcanany, 2024-11-25:
I'm going to eliminate DM from the status change code.  It was only in there in case OLE mode didn't work in some cases.  I'm not worried about that anymore.

For property filters, though, I do need DM (or RM).  That's because I follow model links if a draft file is being processed.  I use that functionality all the time.  I'm wondering if we could check what SE version is installed and choose between DM and RM accordingly.

If you see a hole in that logic, please let me know.  Otherwise I'll add it as a new Issue.

farfilli, 2024-11-26:
I'm not sure on how to handle the duality of DM and RM. 

Another approach would be to open the assembly in Hidden mode to parse the links.

Another approach would be to use that libraries provided with SE to read the SE files without having it installed, but will have to investigate on the capabilities

rmcanany, 2024-11-26:
Latest commit removes DM option for status changing.  Minimally tested at this time.

rmcanany, 2024-12-01:
@farfilli 
Looks like it might be possible to get drawing links from Structured Storage, which would eliminate the need for DM on property filters.  See below.  Come to think of it, it might also allow us to implement the `|R1` syntax elsewhere.

I couldn't figure out how to get data from the OLE Stream.  Any ideas where I could find some documentation or examples to get started?

[image]

farfilli, 2024-12-02:
That part seems to be an OLElink container. I asked ChatGPT how to decode it, and it seems that it's just a matter of parsing the data and recognizing the links.
Here is the [link](https://chatgpt.com/share/674d6baf-2a0c-800d-8770-d73465944eb4) to the conversation (the first part is in Italian so scroll down to the middle for English)

rmcanany, 2024-12-02:
Makes sense about parsing the data, but I'm wondering how to *get* it.  I couldn't find any detailed documentation or examples online.

I guess it would look something like this.

```
dsiStream = tmpStorage.GetStream("K4teagxwOttdbfkuIaamtae3Ie")
co = dsiStream.AsOLEPropertiesContainer

OLEProp = co.Properties.First(Function(Proper) Proper.PropertyName.ToLower = PropertyNameEnglish.ToLower)

```

I can see the `cf` object using a breakpoint.  An OLE stream is present (not shown in the image), but I couldn't find the data it holds.

[image]

farfilli, 2024-12-02:
That stream is not an OLEProperties container therefore you need to read the bytes and covert to text values.
In the ChatGPT conversation I made there are some examples

farfilli, 2024-12-02:
I got to the stream with this code

```
        Dim fs = New FileStream(FullName, FileMode.Open, FileAccess.ReadWrite)
        Dim cfg As CFSConfiguration = CFSConfiguration.SectorRecycle Or CFSConfiguration.EraseFreeSectors
        Dim cf As CompoundFile = New CompoundFile(fs, CFSUpdateMode.Update, cfg)

        Dim A As CFStorage = cf.RootStorage.GetStorage("JSite18446")
        Dim B As CFStream = A.GetStream(ChrW(1) & "Ole")
```

rmcanany, 2024-12-02:
That helps a lot.  Looks like I can get the Byte Array with `B.GetData()`.  I'll play around with it.  Thank you!

rmcanany, 2024-12-02:
Another question... Do you know how to get a list of Storage items?  In this case, I'd like to find all entries named `JSite*`.  

I saw several questions on the openmcdf issues page.  It seems to involve the use of 'tmpStorage.VisitEntries()`, but I don't quite get how to use it.

rmcanany, 2024-12-02:
Never mind the last question.  I think I got it.
`tmpStorage.VisitEntries(Sub(item) dsiStorages.Add(item.Name), recursive:=False)`

farfilli, 2024-12-02:
Version 3.0.0 may have a specific method to list all the sites

farfilli, 2024-12-03:
With this code i'm able to decode the link:
```
    Private Sub FindOleLinks(cf As CompoundFile)

        Dim A As CFStorage = cf.RootStorage.GetStorage("JSite18446")
        Console.WriteLine(GetOleLinkFromStorage(A))

    End Sub

    Private Function GetOleLinkFromStorage(CFStorage As CFStorage) As String

        Dim ST2 = ""

        Dim B As CFStream = CFStorage.GetStream(ChrW(1) & "Ole")
        Dim D = B.GetData()
        Dim ST = Encoding.Unicode.GetString(D)
        Dim B_Start = ST.IndexOf(ChrW(3))
        Dim B_End = ST.IndexOf(ChrW(1))
        ST2 = ST.Substring(B_Start + 1, B_End - B_Start - 1)

        Return ST2

    End Function
```

rmcanany, 2024-12-03:
Slick!  I'll give it a try.  Thank you!

rmcanany, 2024-12-10:
I got the link extraction working I think.  It's not being used anywhere yet.  I'm working on that now in `TaskCheckPartNumberDoesNotMatchFilename`.  Not sure many people use that, but it's an easy place for me to start.

rmcanany, 2024-12-16:
@farfilli @ChrisClems 

I've got this sort of working.  It's quite cumbersome accessing methods in `UtilsCommon`.  I'm wondering if it might be better to create a new class, `StructuredStorageManager` with an API similar to `DesignManager`.

That would let me do stuff like this.

```
Dim SMApp As New StructuredStorageManager.Application

Dim SMDoc As StructuredStorageManager.Document = SMApp.Open(FullName)

Dim LinkedDocuments = SMDoc.LinkedDocuments

SMApp.Quit()
```

Not sure if the effort is worth it.  What do you guys think?  Not sure how to do it, either.  Does this seem like a reasonable start?

```
Public Class StructuredStorageManager

    Public Class Application
        Public Sub New()
        End Sub

        Public Function Open(FullName as String) As Document
            Return New Document(FullName as String)
        End Sub

    End Class

    Public Class Document
        Private fs As FileStream
        Private cf As CompoundFile

        Public Sub New(FullName as String)
            Me.fs = New FileStream(FullName, FileMode.Open, FileAccess.ReadWrite)
            Me.cf = New CompoundFile(fs, CFSUpdateMode.Update, cfg)
        End Sub

    End Class

End Class
```

ChrisClems, 2024-12-16:
> @farfilli @ChrisClems
>
> I've got this sort of working. It's quite cumbersome accessing methods in `UtilsCommon`. I'm wondering if it might be better to create a new class, `StructuredStorageManager` [...]

Sorry, I've had my head buried in another project. I missed the early parts of this discussion.

I've never done any interaction with Design Manager so I'm not at all familiar with the API. My gut reaction was to implement an adapter pattern so DM and RM classes could be accessed through a common interface, but I don't see how that would work when you're referencing one or the other, never both. If I really sat down and tried to put some solutions on paper, it would probably end up being something with some messy reflection and gratuitous use of the dynamic keyword to get it to compile. I'm not sure VB has an equivalent.

I really like the idea of building out a new class through which to access structured storage. That's also on my list for EdgeSharp after I can find time to do some reverse engineering on the structured storage stuff, but my workload has pushed that project way out. I like your idea of mimicking the current API. That will make the structured storage access code much more readable.

rmcanany, 2024-12-16:
Hey Chris,

Really appreciate the feedback.  I'm going to give it a try.  

Part of the thinking is if I get everything I need for the program, mainly properties (which Francesco has already done) and links (which I'm working on), I can do away with DM and RM altogether.

farfilli, 2024-12-17:
I agree, that a class for Structured Storage would be useful.

In the Solid Edge SDK Advanced folder there is a folder called "readers" that contains some DLLs that could be useful for grabbing links and other stuff without Solid Edge installed. I used it in the past, and the documentation is very poor, but in case there is a show-stopper with the OpenMCDF, it could be a solution.

rmcanany, 2024-12-17:
OK, cool.  Appreciate the tip on the SDK.  I'll check it out.

rmcanany, 2024-12-18:
@farfilli 

I'm implementing the code that deals with OLE property types.  For `int` and `double`, it appears the current code automatically changes the type of the property itself to match the value supplied.

I can see how that might be useful, but wonder if a separate method might be better.  I might be missing something.  What do you think?

```
Case = VTPropertyType.VT_I4

    If PropertyValue.ToString = Int(PropertyValue).ToString Then
        Try
            OLEProp.Value = CType(PropertyValue, Integer)
        Catch ex As Exception
            Return False
        End Try

    Else

        Dim newPropertyId As UInteger = OLEProp.PropertyIdentifier
        Dim tmpName = OLEProp.PropertyName
        co.UserDefinedProperties.RemoveProperty(OLEProp.PropertyIdentifier)

        Try

            Dim userProperties = co.UserDefinedProperties

            userProperties.PropertyNames(newPropertyId) = tmpName
            OLEProp = userProperties.NewProperty(VTPropertyType.VT_R8, newPropertyId)
            OLEProp.Value = CType(PropertyValue, Double)
            userProperties.AddProperty(OLEProp)

        Catch ex As Exception
            Return False

        End Try

    End If

```

rmcanany, 2024-12-20:
@farfilli 
A couple other questions for you...

In this code, it looks like it's accounting for some leading or trailing text in the property name.  I don't see that on my system.  Could you explain what's going on?
```
ElseIf (FunnyList.Contains(PropertyNameEnglish)) And (PropertySet.ToLower = "system") Then
    dsiStream = tmpStorage.GetStream("Rfunnyd1AvtdbfkuIaamtae3Ie")
    co = dsiStream.AsOLEPropertiesContainer

    OLEProp = co.Properties.FirstOrDefault(Function(Proper) Proper.PropertyName.ToLower Like "*" & PropertyNameEnglish.ToLower & "*")
```

In case you're using Structured Storage in other projects, I noticed a typo you might want to know about.  The DocumentSummaryInformation property name prefix is `PIDDSI_`, not `PIDSI_`.
```
ElseIf (DSIList.Contains(PropertyNameEnglish)) And (PropertySet.ToLower = "system") Then
    dsiStream = tmpStorage.GetStream("DocumentSummaryInformation")
    co = dsiStream.AsOLEPropertiesContainer

    OLEProp = co.Properties.First(Function(Proper) Proper.PropertyName = "PIDSI_" & PropertyNameEnglish.ToUpper)
```

farfilli, 2024-12-20:
It was to avoid typing the PIDDSI text that is not ideal in the user interface, you can change the first * with PIDDSI_
The .ToLower was to not consider how the user typing the property itself

rmcanany, 2024-12-20:
Got it.  Thank you!

rmcanany, 2025-01-15:
DesignManager has been replaced throughout.  Most places, SSDoc is used instead.  In a couple of places, notably in TLA Bottom Up search, I replaced it with RevisionManager, because of the Where Used functionality.  Needs more testing, but it at least compiles.

If anyone knows a reason not to use RM, besides the fact it's deprecated (like for almost 10 years now), please let me know.

rmcanany, 2025-01-18:
Closing as complete.

closed 2025-01-18 by rmcanany