rmcanany/SolidEdgeHousekeeper #179: Move change status from configuration to a proper task

farfilli, 2024-11-22:
The need is to change the status of files only.
With the actual method, you can't and are forced to activate at least one task on the list.
Creating a proper change status task would also allow to use the direct edit (by openMCDF) to change the status in a blazing fast way

rmcanany, 2024-11-25:
I see your point.  One advantage of manipulating the Status in the configuration is that someone can change a file to Available, say for updating drawing views, then change it back.  To me, that is still needed.  

However for the case you describe, it isn't.  Let me take that one.  I already planned on changing DM to OLE on the configuration side.  I can easily add a Status Change task that is compatible.  Sound OK?

farfilli, 2024-11-25:
What if we implement the status change in the edit properties task?

rmcanany, 2024-11-25:
I'm open to it, but wonder if a separate task might be cleaner.  

For one thing, Edit Properties can run in SE mode -- you can't change from a Read Only status that way.  We could check for that condition, but it sounds convoluted.  

Also, as you know, two properties must be set for a status change.  We could deal with that, too,  But would then probably need to check that the user isn't already setting both on their own.

One more thing.  I kinda think a status change should run separately from other tasks.  I could easily add a flag for that.  That way I could run it in OLE mode without further checks.

Anyway, those are my thoughts at the moment.  Comments welcome!

farfilli, 2024-11-26:
A separate task would be clearer for sure, I agree to go for it.

One more thing to consider about status change is the user that commits the change. I'm not sure what property handles that value, will have to check

rmcanany, 2024-11-27:
Added TaskSetDocumentStatus.  [Help topic](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/master/README.md#set-document-status), FYI.  Minimally tested so far.  Let me know if you run into anything.

I didn't add Author, Change Date, or any other property updates, but can easily do so if needed.

farfilli, 2024-11-28:
Changing the status with Solid Edge creates these unnamed two properties that represent the Author and DateTime of the Status

[image]

In a new file, those two properties are not present therefore a check is needed before update\create them

Perhaps the UI could have a field where manually fill the user to use to sign the status change

rmcanany, 2024-11-28:
I like the idea of having the user choose the property fields manually.  I'll give that a try.

rmcanany, 2024-11-30:
Added updating of additional properties.  (RMB any textbox to select properties from your Favorites list.)

Note, if the task list already has TaskSetDocumentStatus, it will have to be removed and re-added.  There are some new Property fields, which the previously saved task will not have.

rmcanany, 2025-01-15:
I changed my mind about making changing Status a stand-alone task.  It can run with any other Task that supports the `Run without SE` mode.  

Regarding the related fields, as you note the Edit Properties can be used for those.

rmcanany, 2025-01-18:
Closing as complete.

closed 2025-01-18 by rmcanany