rmcanany/SolidEdgeHousekeeper #182: Will there be a Chinese version?

chjchen, 2024-11-30:
A very good tool, will there be a Chinese version in the future?

farfilli, 2024-11-30:
The tool is not written for localization, it would need a big effort to do it.
Let's see if it's ok for Robert to do such important improvements. I'll do my part of course.

chjchen, 2024-11-30:
Thank you,I hope to be able to use it soon.

farfilli, 2024-11-30:
In case you will have to contribute with the Chinese translation

rmcanany, 2024-11-30:
Thank you for the suggestion!  

There are no plans for that right now, unfortunately.  However, if someone wanted to take it on, I'm sure a lot of users would welcome the improvement!

farfilli, 2024-12-02:
This [link ](https://learn.microsoft.com/en-us/windows/apps/design/globalizing/use-mat)contains the instructions on how to implement multi-lingual support.
Once the application is structured to support it, adding a language is an easy task.
However, the [first step](https://learn.microsoft.com/en-us/windows/uwp/app-resources/localize-strings-ui-manifest) to perform could be a big work, therefore we have to consider and plan it in case.

closed 2024-12-03 by chjchen