rmcanany/SolidEdgeHousekeeper #183: Property Filter

FERNANDO-DALLAGNOLO, 2024-12-03:
Hi,

How can I configure the filter so that files that have the word OXICORTE in their name do not appear in the results?

[image]

farfilli, 2024-12-03:
You need to insert the wildcard "*" in front of "- OXIXORTE.dft"

FERNANDO-DALLAGNOLO, 2024-12-03:
> You need to insert the wildcard "*" in front of "- OXIXORTE.dft"

In the wildcard I put *D2303011* to find only the manufactured files and not include standardized items such as screws, washers and pins. But in these results I find the 2D files that contain OXICORTE in their filename and I don't want these files to appear in the results.
The way you suggested works but it's not what I need because this way it only finds the files that contain OXICORTE in the name.
for example:
there are two files
D2303011001.dft
D2303011001 - OXICORTE.dft

I want to filter and find only the first one, in this case:
D2303011001.dft

farfilli, 2024-12-03:
[image]

FERNANDO-DALLAGNOLO, 2024-12-03:
it doesn't work, it still finds all the files
[image]

farfilli, 2024-12-03:
There must be a bug cause the "is not" always revert to "contains"; @rmcanany can you take a look at this situation?

FERNANDO-DALLAGNOLO, 2024-12-03:
sorry, but it's not working
I thought it could be the space or the hyphen but it's not
I've done several tests, and it works as you indicated when I filter by the letter W for example, the app eliminates the results composed of D2303011W01
D2303011W02
[image]
there are 440 results without using a filter
[image]
and 158 results with the filter on the letter W
[image]

rmcanany, 2024-12-03:
One other thing to check.  For Draft files there is an option to also check properties of the model files it contains.  It's on by default, but might be worth turning off for this case.  The option is on the **Configuration Tab -- General Page**.  Please try that and let me know how it goes.

[image]

rmcanany, 2024-12-03:
Thinking more about it, you might also have to enable the option `include the draft file itself`.

FERNANDO-DALLAGNOLO, 2024-12-04:
sorry, but it's not working

FERNANDO-DALLAGNOLO, 2024-12-04:
I made some test files, in total there are 10 files
[image]
when applying the filter to eliminate from the results the files that contain the word TEST in the name and the result is this
[image]
[image]
it did not delete the dft files
[image]

rmcanany, 2024-12-04:
Sounds like a bug.  I'll dig into it this afternoon.  

In the meantime, try to uncheck `Property Filter -- Include Draft file model documents in search`.

rmcanany, 2024-12-04:
Got it.  Thank you.

It seems to work with options set like this: `Exclude models`, `Include drafts`.  See images below.  

In the test case, the two drafts with TESTE in the name contain models without that text in their own names.  That's where the match is coming from.  

In the original case you presented, it was either that, and/or the fact `.dft` was part of the filter.  No model names end in `dft`, so you would get a match on that basis, too.

Anyway, my test results are below.  Please let me know if it doesn't work for you with those settings.  It's still possible there's a bug somewhere.

All files:
[image]

Property filter:
[image]

Property search options:
[image]

Results:
[image]

FERNANDO-DALLAGNOLO, 2024-12-04:
That way it worked, thanks

closed 2024-12-04 by rmcanany