rmcanany/SolidEdgeHousekeeper #184: Exclude family of parts master

farfilli, 2024-12-04:
A user asked to be able to exclude the FOP master when retrieving files from a top-level assembly file.
It could be an option for the configuration or a filter that is more useful all around. I'm not sure how to recognize a link as master of a FOP

rmcanany, 2024-12-04:
Let me take this one.  That TLA code is a hairball.  

It would be helpful if they could provide a simple example.  Also I would need to know if they're running in `Bottom up` or `Top down` mode.

For the former mode, it should be checking already.  It's in `UtilsTopLevelAssembly.FindLinksBottomUp()`.  Not positive it's working correctly, however.  

For the latter mode, I don't see a check in `FindLinksTopDown()`.

rmcanany, 2025-01-15:
This should be working now, both for Top Down and Bottom Up searches.  I can put together a preview version out on my Google Drive if your guy wants to try it out.

rmcanany, 2025-01-18:
Closing as complete.

closed 2025-01-18 by rmcanany