rmcanany/SolidEdgeHousekeeper #187: Save As - filename

ih0nza, 2025-01-09:
Hello, I have similar thing: For documents, we use unique file names, and the document number is filled in the file properties as an attribute. For batch exports to DWG, PDF, and STEP, we currently use the Batch.exe program provided with Solid Edge, which we have customized to create a file names like "(DocumentNumber) (Filename).pdf" during export. Is it possible to define in Housekeeper that this "DocumentNumber" is added before the file name? Thank you.

[image]

_Originally posted by @ih0nza in https://github.com/rmcanany/SolidEdgeHousekeeper/issues/174#issuecomment-2579579710_

rmcanany, 2025-01-15:
Closing as complete in the upcoming version.  Thank you!

rmcanany, 2025-01-15:
Closing for real.

closed 2025-01-15 by rmcanany