rmcanany/SolidEdgeHousekeeper #188: Integer vs Boolean values stored in SS OLEProps

farfilli, 2025-01-14:
Numeric properties in SS are managed in a "Weird" way, I assume this is to save some bites in the file streams.

If the number to store is an integer the property type is VT_I4; instead, if the number to store is double the property type is VT_R8

In this line, the number is always converted to an integer and that will lead to a decimal loss in case the value to store is double
https://github.com/rmcanany/SolidEdgeHousekeeper/blob/92a35dd374712c0c8d904bb2170471eb1af37af6/My%20Project/HCStructuredStorageDoc.vb#L1034

There is no method to change the property type therefore the only way is to delete the property and recreate it retaining its id (in my original code this technique was used)

rmcanany, 2025-01-14:
I see what you mean.  I just did a test in SE.  I created two new custom properties of type `Number`.  One had a decimal point and the other didn't.  Checking in SSView, the first was an `int`, the other was a `double`.  I went back in SE and added a decimal to the first and removed it from the second.  Back in SSView, the types had switched.  

I'll fix that today.  I appreciate you keeping an eye on things, by the way.

rmcanany, 2025-01-15:
That should be working now.  If you get a chance to try it, please let me know how it goes.  

What's not working is `Number` type properties on the SE side.  Is there some sort of trick to it?  In `TaskEditProperties`, I do `Prop.Value = CInt("1")` for a `VT_I4` property and get something like `1448905303`.  Huh?

farfilli, 2025-01-15:
Converting a string to an integer could be tricky; did you try using a different syntax?
`OLEProp.Value = CType(PropertyValue, Integer)`

Or event better declare an intermediate variable as `Integer `and use it to assign the converted value? In this way you can check the conversion before pass the value

farfilli, 2025-01-15:
One more hint, in the code you use `Int32` which is a different beast than `Integer` perhaps this is the issue? Or do you use Int32 just as a tag?

rmcanany, 2025-01-15:
Tried it both ways with no luck.  I tried the following with a breakpoint on the second line.  `i` is set to `2`, but it's still not working.  I'll keep playing with it.
```
Dim i As Integer = CInt("2")
Prop.Value = i
```

rmcanany, 2025-01-24:
I see SE 2024 Update 11 fixed an API bug: PR 10852832: Application Crash Due to Solid Edge Status Setting

I'm going to try to update next week and see if it fixes my issue with Int and Double.

rmcanany, 2025-02-16:
Still having issues with numeric properties.  Sent the following to our VAR on Friday.

==========================================
Problem setting the value for a numeric property via API

Running SE 2024 MP11 and having trouble setting the value of a numeric property using the API.  I have a test case prepared.  The relevant part is below.  Note the console output says Prop.Type is Int32, but Prop.Value is 0.29 (a Double).

CODE

    Dim PropertySetName As String = "MechanicalModeling"
    Dim PropName As String = "Poisson's Ratio"
    Dim PropValue as Double = 3.1415

    PropertySets = SEDoc.Properties
    PropertySet = PropertySets.Item(PropertySetName)
    Prop = PropertySet.Item(PropName)

    Console.WriteLine(String.Format("PropertySetName '{0}'", PropertySetName))
    Console.WriteLine(String.Format("PropName '{0}'", PropName))
    Console.WriteLine(String.Format("Prop.Type '{0}'", Prop.Type.GetType.Name))
    Console.WriteLine(String.Format("Prop.Value '{0}'", CStr(Prop.Value)))

    Prop.Value = PropValue

    If Prop.Value = PropValue Then  ' <---------- Throws an Exception
        Console.WriteLine("Success")
    End If

CONSOLE OUTPUT

    PropertySetName 'MechanicalModeling'
    PropName 'Poisson's Ratio'
    Prop.Type 'Int32'
    Prop.Value '0.29'

rmcanany, 2025-02-16:
Here's the source code if you want to mess with it.  Note, there are two files -- Module1.vb and OleMessageFilter.vb.

Module1.vb
```
Module Module1
    <STAThread>
    Public Sub Main(ByVal args() As String)
        Dim SEApp As SolidEdgeFramework.Application = Nothing
        Dim SEDoc As SolidEdgeFramework.SolidEdgeDocument = Nothing
        Dim PropertySets As SolidEdgeFramework.PropertySets
        Dim PropertySet As SolidEdgeFramework.Properties
        Dim Prop As SolidEdgeFramework.Property

        Dim Filename = "C:\data\CAD\scripts\TestNumericProperty\Part1.par"

        If Not IO.File.Exists(Filename) Then
            Dim s As String = String.Format("File name is hard-coded in Module1.vb{0}{0}", vbCrLf)
            s = String.Format("{0}'{1}' not found{2}", s, Filename, vbCrLf)
            MsgBox(s)
            Exit Sub
        End If

        OleMessageFilter.Register()

        Console.WriteLine("Starting Solid Edge")
        SEApp = Microsoft.VisualBasic.CreateObject("SolidEdge.Application")
        SEApp.Visible = True
        Console.WriteLine(String.Format("Opening {0}", Filename))
        SEDoc = DirectCast(SEApp.Documents.Open(Filename), SolidEdgeFramework.SolidEdgeDocument)

        SEDoc.Activate()

        Dim PropertySetName As String = "MechanicalModeling"
        Dim PropName As String = "Poisson's Ratio"
        Dim PropValue As Double = 3.1415

        PropertySets = SEDoc.Properties
        PropertySet = PropertySets.Item(PropertySetName)
        Prop = PropertySet.Item(PropName)

        Console.WriteLine(String.Format("PropertySetName '{0}'", PropertySetName))
        Console.WriteLine(String.Format("PropName '{0}'", PropName))
        Console.WriteLine(String.Format("Prop.Type '{0}'", Prop.Type.GetType.Name))
        Console.WriteLine(String.Format("Prop.Value '{0}'", CStr(Prop.Value)))

        Prop.Value = PropValue

        If Prop.Value = PropValue Then ' <-------------- Throws an exception
            Console.WriteLine("Success")
        End If

        SEDoc.Save()
        SEApp.DoIdle()
        SEDoc.Close()
        SEApp.DoIdle()

        Console.WriteLine("SEApp.Quit()")
        SEApp.Quit()

        OleMessageFilter.Revoke()

    End Sub

End Module

```

OleMessageFilter.vb

```

Imports System.Runtime.InteropServices
Imports System.Threading

Public Class OleMessageFilter
    Implements IOleMessageFilter

    Public Shared Sub Register()
        Dim newFilter As IOleMessageFilter = New OleMessageFilter()
        Dim oldFilter As IOleMessageFilter = Nothing
        Dim iRetVal As Integer

        If (Thread.CurrentThread.GetApartmentState() = ApartmentState.STA) Then
            iRetVal = CoRegisterMessageFilter(newFilter, oldFilter)
        Else
            Throw New COMException("Unable to register message filter because the current thread apartment state is not STA.")
        End If

    End Sub

    Public Shared Sub Revoke()
        Dim oldFilter As IOleMessageFilter = Nothing
        CoRegisterMessageFilter(Nothing, oldFilter)
    End Sub

    Private Function HandleInComingCall(ByVal dwCallType As Integer,
                                        ByVal hTaskCaller As System.IntPtr,
                                        ByVal dwTickCount As Integer,
                                        ByVal lpInterfaceInfo As System.IntPtr) _
                                        As Integer Implements IOleMessageFilter.HandleInComingCall

        Return SERVERCALL.SERVERCALL_ISHANDLED
    End Function

    Private Function RetryRejectedCall(ByVal hTaskCallee As System.IntPtr,
                                       ByVal dwTickCount As Integer,
                                       ByVal dwRejectType As Integer) _
                                       As Integer Implements IOleMessageFilter.RetryRejectedCall

        If dwRejectType = SERVERCALL.SERVERCALL_RETRYLATER Then
            Return 99
        End If

        Return -1
    End Function

    Private Function MessagePending(ByVal hTaskCallee As System.IntPtr,
                                    ByVal dwTickCount As Integer,
                                    ByVal dwPendingType As Integer) _
                                    As Integer Implements IOleMessageFilter.MessagePending

        Return PENDINGMSG.PENDINGMSG_WAITDEFPROCESS
    End Function

    <DllImport("Ole32.dll")>
    Private Shared Function CoRegisterMessageFilter(ByVal newFilter As IOleMessageFilter,
                                                    ByRef oldFilter As IOleMessageFilter) _
                                                    As Integer
    End Function
End Class

Enum SERVERCALL
    SERVERCALL_ISHANDLED = 0
    SERVERCALL_REJECTED = 1
    SERVERCALL_RETRYLATER = 2
End Enum

Enum PENDINGMSG
    PENDINGMSG_CANCELCALL = 0
    PENDINGMSG_WAITNOPROCESS = 1
    PENDINGMSG_WAITDEFPROCESS = 2
End Enum

<ComImport(),
    Guid("00000016-0000-0000-C000-000000000046"),
    InterfaceTypeAttribute(ComInterfaceType.InterfaceIsIUnknown)>
Interface IOleMessageFilter
    <PreserveSig()>
    Function HandleInComingCall(ByVal dwCallType As Integer,
                                ByVal hTaskCaller As IntPtr,
                                ByVal dwTickCount As Integer,
                                ByVal lpInterfaceInfo As IntPtr) _
                                As Integer

    <PreserveSig()>
    Function RetryRejectedCall(ByVal hTaskCallee As IntPtr,
                               ByVal dwTickCount As Integer,
                               ByVal dwRejectType As Integer) _
                               As Integer

    <PreserveSig()>
    Function MessagePending(ByVal hTaskCallee As IntPtr,
                            ByVal dwTickCount As Integer,
                            ByVal dwPendingType As Integer) _
                            As Integer
End Interface

```

rmcanany, 2025-03-25:
Converted to a PR, but I didn't get the number from our VAR.  Guess I'll just leave this open.

rmcanany, 2025-04-13:
Opened https://github.com/rmcanany/SolidEdgeHousekeeper/issues/208 for the API issue.

Closing as complete.

rmcanany, 2025-04-13:
Closing for real.

closed 2025-04-13 by rmcanany